volumio/Volumio2 #425: [RPi] Volumio2 :: Stuck after boot

jo3rg, 2016-05-19:
Hi,
I'm struggling with my raspberry RPi setup.
Somehow it's getting stuck after booting when connected to the network via WiFi:
------------------------------------------>
volumio@volumio:~$ tail -f /var/log/volumio.log 
2016-05-19T15:34:04.980Z - info: -------------------------------------------
2016-05-19T15:34:05.094Z - info: -----            Volumio2              ----
2016-05-19T15:34:05.104Z - info: -------------------------------------------
2016-05-19T15:34:05.123Z - info: -----          System startup          ----
2016-05-19T15:34:05.129Z - info: -------------------------------------------
2016-05-19T15:34:05.467Z - info: Loading plugins from folder /volumio/app/plugins/
2016-05-19T15:34:05.647Z - info: Loading plugins from folder /data/plugins/
2016-05-19T15:34:05.687Z - info: Loading plugin "system"...
2016-05-19T15:34:06.333Z - info: Loading plugin "i2s_dacs"...
2016-05-19T15:34:06.441Z - info: Loading plugin "network"...
2016-05-19T15:34:07.173Z - info: Loading plugin "networkfs"...
2016-05-19T15:34:07.354Z - info: Loading plugin "services"...
2016-05-19T15:34:07.413Z - info: Loading plugin "alsa_controller"...
2016-05-19T15:34:10.334Z - info: Loading plugin "volumiodiscovery"...
2016-05-19T15:34:11.057Z - info: Applying required configuration parameters for plugin volumiodiscovery
2016-05-19T15:34:11.488Z - info: CoreCommandRouter::executeOnPlugin: system , registerCallback
2016-05-19T15:34:11.509Z - info: Loading plugin "upnp"...
2016-05-19T15:34:11.587Z - info: [1463672051561] Starting Upmpd Daemon
2016-05-19T15:34:11.613Z - info: CoreCommandRouter::executeOnPlugin: system , registerCallback
2016-05-19T15:34:11.621Z - info: Loading plugin "airplay_emulation"...
2016-05-19T15:34:11.693Z - info: [1463672051673] Starting Shairport Sync
2016-05-19T15:34:11.731Z - info: Loading plugin "dirble"...
2016-05-19T15:34:17.883Z - info: Applying required configuration parameters for plugin dirble
2016-05-19T15:34:17.936Z - info: CoreCommandRouter::volumioAddToBrowseSources[object Object]
2016-05-19T15:34:17.955Z - info: [1463672057939] CoreMusicLibrary::Adding element WebRadio
2016-05-19T15:34:17.963Z - info: CoreCommandRouter::BroadCastMessage pushBrowseSources
2016-05-19T15:34:18.034Z - info: Loading plugin "alarm-clock"...
2016-05-19T15:34:18.771Z - info: Loading plugin "albumart"...
2016-05-19T15:34:18.873Z - info: Plugin example_plugin is not enabled
2016-05-19T15:34:18.894Z - info: Loading plugin "mpd"...
2016-05-19T15:34:21.283Z - info: Loading plugin "spop"...
2016-05-19T15:34:23.233Z - info: [1463672063227] ControllerSpop::loadTracklistFromDB
2016-05-19T15:34:23.243Z - info: Loading Spop tracklist from DB...
2016-05-19T15:34:23.446Z - info: Loading plugin "updater_comm"...
2016-05-19T15:34:26.333Z - info: Plugin mpdemulation is not enabled
2016-05-19T15:34:26.344Z - info: Loading plugin "websocket"...
2016-05-19T15:34:26.813Z - info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
2016-05-19T15:34:26.833Z - info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
2016-05-19T15:34:26.840Z - info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
2016-05-19T15:34:26.846Z - info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
2016-05-19T15:34:26.963Z - info: BOOT COMPLETED
2016-05-19T15:34:27.308Z - info: [1463672067291] CoreStateMachine::resetVolumioState
2016-05-19T15:34:27.318Z - info: [1463672067312] CoreStateMachine::getcurrentVolume
2016-05-19T15:34:27.328Z - info: CoreCommandRouter::volumioRetrievevolume
2016-05-19T15:34:27.412Z - info: [1463672067395] CoreStateMachine::updateTrackBlock
2016-05-19T15:34:27.453Z - info: [1463672067416] CorePlayQueue::getTrackBlock
2016-05-19T15:34:28.120Z - info: ------------ TEST ENVIRONMENT DETECTED ---------------
2016-05-19T15:34:28.773Z - info: SpopD Daemon Started
2016-05-19T15:34:28.818Z - info: MPD running with PID448 ,establishing connection
2016-05-19T15:34:28.927Z - info: Setting Device type: Raspberry PI
2016-05-19T15:34:29.073Z - info: [1463672069066] ControllerSpop::pushError(Error: Error reading DB: NotFoundError: Key not found in database [tracklist])
2016-05-19T15:34:30.138Z - info: [1463672070129] VolumeController:: Volume=40 Mute =false
2016-05-19T15:34:30.167Z - info: [1463672070151] CoreStateMachine::updateVolume40
2016-05-19T15:34:30.183Z - info: [1463672070171] CoreStateMachine::pushState
2016-05-19T15:34:30.198Z - info: [1463672070187] CoreStateMachine::getState
2016-05-19T15:34:30.212Z - info: CoreCommandRouter::volumioPushState
2016-05-19T15:34:30.222Z - info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
2016-05-19T15:34:30.257Z - info: [1463672070247] InterfaceWebUI::pushState
2016-05-19T15:34:30.407Z - info: Pushing Favourites {"service":null,"uri":null,"favourite":false}
2016-05-19T15:34:31.085Z - info: 
2016-05-19T15:34:31.484Z - info: mDNS: Found device Volumio
2016-05-19T15:34:31.815Z - info: mDNS: Found device Volumio
2016-05-19T15:34:32.835Z - info: Upmpdcli Daemon Started
2016-05-19T15:34:32.874Z - info: [1463672072839] Shairport-Sync Started
------------------------------------------>
I'm using an unmodified RC1 image: volumio-0.861-2016-03-25-pi.img
Strange thing is that this only happens when connected via WiFi. 
If I connect it via Ethernet it works like a charm...
Has anyone similar problems? How can I stop the Shairport Service?

Thanks in advance and regards
Joerg

volumio, 2016-05-19:
That's quite weird... 
Could you do please 

killall node 

and paste the full log of 

sudo journalctl -p 7 

thanks!

jo3rg, 2016-05-19:
here you go:
volumio@volumio:~$ killall node
volumio@volumio:~$ sudo journalctl -p 7
-- Logs begin at Thu 1970-01-01 00:00:13 UTC, end at Thu 2016-05-19 21:12:12 UTC. --
Jan 01 00:00:13 volumio systemd-journal[136]: Runtime journal is using 4.0M (max allowed 30.0M, trying to leave 36.1M free of 
237.1M available → current limit 30.0M).
Jan 01 00:00:13 volumio kernel: Booting Linux on physical CPU 0x0
Jan 01 00:00:13 volumio kernel: Initializing cgroup subsys cpuset
Jan 01 00:00:13 volumio kernel: Initializing cgroup subsys cpu
Jan 01 00:00:13 volumio kernel: Initializing cgroup subsys cpuacct
Jan 01 00:00:13 volumio kernel: Linux version 4.1.19+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng
-1.22.0-88-g8460611) ) #851 Sat Mar 5 22:26:57 GMT 2016
Jan 01 00:00:13 volumio kernel: CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
Jan 01 00:00:13 volumio kernel: CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
Jan 01 00:00:13 volumio kernel: Machine model: Raspberry Pi Model B Rev 2
Jan 01 00:00:13 volumio kernel: cma: Reserved 8 MiB at 0x1dc00000
Jan 01 00:00:13 volumio kernel: Memory policy: Data cache writeback
Jan 01 00:00:13 volumio kernel: On node 0 totalpages: 126976
Jan 01 00:00:13 volumio kernel: free_area_init_node: node 0, pgdat c0839a04, node_mem_map dd7a4000
Jan 01 00:00:13 volumio kernel:   Normal zone: 1116 pages used for memmap
Jan 01 00:00:13 volumio kernel:   Normal zone: 0 pages reserved
Jan 01 00:00:13 volumio kernel:   Normal zone: 126976 pages, LIFO batch:31
Jan 01 00:00:13 volumio kernel: pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Jan 01 00:00:13 volumio kernel: pcpu-alloc: [0] 0 
Jan 01 00:00:13 volumio kernel: Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 125860
Jan 01 00:00:13 volumio kernel: Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 
bcm2708.boardrev=0xe bcm2708.serial=0x53f2b4a3 smsc95xx.macaddr=B8:27:EB:F2:B4:A3 bcm2708_fb.fbswap=1 bcm2708.uart_clock=30000
00 vc_mem.mem_base=0x1fa00000 vc_mem.mem_size=0x20000000  force_turbo=1 dwc_otg.lpm_enable=0  dwc_otg.fiq_enable=1 dwc_otg.fiq
_fsm_enable=1 dwc_otg.fiq_fsm_mask=0x3 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 imgpart=/dev/mmcblk0p2 imgfil
e=/volumio_current.sqsh elevator=noop rootwait smsc95xx.turbo_mode=N
Jan 01 00:00:13 volumio kernel: PID hash table entries: 2048 (order: 1, 8192 bytes)
Jan 01 00:00:13 volumio kernel: Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
Jan 01 00:00:13 volumio kernel: Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
Jan 01 00:00:13 volumio kernel: Memory: 477280K/507904K available (5796K kernel code, 488K rwdata, 1788K rodata, 348K init, 72

volumio, 2016-05-19:
Mmm sorry that's a small part of the log...

apinner, 2016-05-20:
Can you paste the whole log into something like http://pastebin.com/ 

It would make it much easier to look at.

Saiyato, 2016-05-23:
I've had similar issues too, I don't have logs present, since I re-flashed like a million times.
It only seemed to occur after rebooting, either via GUI or using the sudo reboot command.

Terminal outputs (photo, because the pi was unresponsive)
[image]

And number two
[image]

This one is actually more common, ending with the kdb_stub message.

Which logs should I check and post if this happens again?

xipmix, 2016-11-30:
The screen shots show an issue with the broadcom wireless driver which has had some improvements over the last few months. Are you folks still seeing this with version 2.031?

If you are seeing this, try the following after booting up.
```
volumio$ sudo journalctl -p7 >journalctl-p7.txt
```
That will give you a file on disk.  If you reboot with ethernet connected (which seems to work) you should be able to log in from another machine and grab the saved file, e.g.
```
% sftp volumio@volumio
volumio@volumio's password:
Connected to volumio.
sftp> get journalctl-p7.txt
sftp> quit
%
```

closed 2017-06-02 by volumio