volumio/Volumio2 #428: less is missing

durilka, 2016-05-21:
I don't know if it's intentional but "less" is not in /bin.
RC2

apinner, 2016-05-21:
Why would you need it included in Volumio?

durilka, 2016-05-21:
well, because it's very good for log/config viewing. Volumio is of course a out-of-the-box system, but it's still is a thing that can go wrong. And my standard toolkit is find/grep/less/nano for that.

in my case it was a discovery that the configuration template (*conf.tmpl) is always edited and written to the /etc which means one should not edit files in /etc (which is not an obvious thing)

but again, I approach it as a linux system in a first place, not as something that one just plugs into network and then uses the web ui at most.

volumio, 2016-05-23:
Added 
https://github.com/volumio/Build/commit/1b24f5141e925cb00bf96f00aed173450b565517

Only some of the conf files are added to /etc , so you can still edit the other ones... Only the one we need to edit are touched (networking, mpd etc..)

closed 2016-05-23 by volumio