volumio/Volumio2 #433: VOLUMIO2 RC1 LONG PLAY FILE PROBLEM

ycat3, 2016-05-23:
I have some long play(more than 6 hours) m4a files.
No sound will come out halfway, around 40 minutes after playing even though Web GUI is working and count up.
Raspberry Pi 3 with JBL pebbles USB speaker and ODROID C1+ with I2S card show the same problem.
I have verified these m4a files with other platform without problem.
The following link shows the sample m4a file.
https://www.dropbox.com/s/29zxlq4kahcdppr/LongPlaySample.zip?dl=0

klokslagB, 2016-05-24:
I'm playing the file know on my setup. Will edit this comment in 40 minutes.

klokslagB, 2016-05-24:
Hmmm I glad that I waited just a little bit longer at 21:46:XX The music stopped playing.
So I can confirm this bug.

xipmix, 2017-01-30:
Hi, could you close this issue if it is fixed for you now?

closed 2017-06-02 by volumio