volumio/Volumio2 #438: Bauer crossfeed for headphones listening

apinner, 2016-05-26:
On behalf of [Badger](https://volumio.org/forum/member2862.html)
Hi to all!
I'm here for suggest a feature for Volumio 2.
Lots of headphones amplifiers for raspberry are borning and I2S DACs often have headphones out, jack 3.5mm. 
I have a Durio Sound Pro and I listen both speakers and headphones. 
For a better headphones listening should be fine have crossfeed filter, here below Bauer stereophonic-to-binaural DSP.
XBMC has already this feature, could you add it to Volumio 2 easily? 

http://bs2b.sourceforge.net/download.html

Thanks for help and for time spent for Volumio project! 
Thanks a lot!

volumio, 2016-05-26:
Why not... there could be a plugin...

volumio, 2018-03-26:
Done via a plugin

closed 2018-03-26 by volumio

mbrennwa, 2019-06-21:
Where / how do I find the crossfeed plugin?