volumio/Volumio2 #442: Need driver for Soekris I2S DAC

manojo1, 2016-05-27:
Need driver for Soekris I2S DAC. A few have made it work through some tweaks with the previous version of Volumio and pi2. Would be great if the Volumio2 can provide the driver for Pi3 ti make Soekris DAC work with I2S out of the box.

volumio, 2016-05-28:
Could you paste the discussions\thread links for this here?

manojo1, 2016-05-28:
https://volumio.org/forum/soekris-ladder-dac-t2553.html

https://volumio.org/forum/volumio-pi3-soekris-t4234.html

http://www.diyaudio.com/forums/vendors-bazaar/259488-reference-dac-module-discrete-r-2r-sign-magnitude-24-bit-384-khz.html

johnyn6, 2016-07-30:
I have Soekris I2S DAC connected with Odriod C1+/ C2  over IS2 out and DAC is working with odroiddac driver up to 192KHz PCM. Higher sample rates and DSD are very noisy and volume level is wrong. Driver for Soekris DAC would be great.

volumio, 2016-09-16:
Could you please try to use the RaspiDACv3 and report if that works?

closed 2017-06-02 by volumio