volumio/Volumio2 #444: Hifiberry Amp+: No volume adjustment

Turbofoehn, 2016-05-29:
Hi everyone!
I am running Volumio on a Raspberry 2 with a Hifiberry Amp+. Unfortunately, setting the volume has no effect: All music blasts at the loudest level.
Curiously, it works when streaming via Airplay.

apinner, 2016-06-29:
@Turbofoehn can you confirm if this is still an issue in RC2?

ElLoco1, 2016-06-30:
I had it before on RC1 too but playing with the different mixer controls and changing DAC models solved it at a certain moment.
Under RC2 I have the issue other people are also facing where the Output Device is stuck on 'audio jack' and mixer control is greyed out. So sound comes out, but at 100% volume.

Will the Hifiberry Amp+ be added as a separate DAC model in the future? As far as I know this has hardware volume control too.

apinner, 2016-06-30:
Yeah we are aware of the audio jack and mixer issue.

The HifiBerry Amp+ is already added as a seperate DAC model.

https://github.com/volumio/Volumio2/blob/fdd5fd661a0766cece06249e5ebf669f82af4ed8/app/plugins/system_controllers/i2s_dacs/dacs.json#L10

Turbofoehn, 2016-07-03:
@apinner The issue is still very much present with RC2, my poor ears. 🙉

I don't know whether this is relevant, but the config file you linked contains `"mixer":"Digital"` for the amp, while [this page](https://www.hifiberry.com/2014/12/guide-configure-volumio-1-51-for-hifiberry-ampamp/) recommends using `mixer_type     "hardware"`

apinner, 2016-07-04:
@Turbofoehn are you comfortable in an SSH window? Do you know how to edit that file and try making that change?

dragotin, 2016-07-04:
Can confirm this issue with HifiBerry AMP on RC2. I did not succeed to adjust values according to https://www.hifiberry.com/2014/12/guide-configure-volumio-1-51-for-hifiberry-ampamp/ but alsamixer callled on the commandline works to adjust the volume.

mundodisco8, 2016-07-05:
Same here. Raspberry pi 2 + Hifiberry Amp on RC2 (the 30th of June one). Changing the volume rocker in the GUI has no effect on the volume levels.

Amixer only shows one control, and I can't change the volume

```
Simple mixer control 'PCM',0
  Capabilities: pvolume pvolume-joined pswitch pswitch-joined
  Playback channels: Mono
  Limits: Playback -10239 - 400
  Mono: Playback -5025 [49%] [-50.25dB] [on]
```

Running alsamixer has no effect on the volume levels. Could it be that Volumio2 is not detecting the DAC correctly? should a different mixer control appear as output of amixer?

Turbofoehn, 2016-07-05:
@mundodisco8 Did you select the Hifiberry as output device in alsamixer? I can’t verify it right now, but I recall being able to manually set the volume there.

overflowed, 2016-07-23:
By default it seems the mixer is set to Digital, however Digital is not present when I change in the gui the mixer to Master I can control the volume. Is it just changing 

`    {"id":"hifiberry-amp","name":"Hifiberry Amp","overlay":"hifiberry-amp","alsanum":"1","mixer":"Digital","modules":"","script":""},`
to
`    {"id":"hifiberry-amp","name":"Hifiberry Amp","overlay":"hifiberry-amp","alsanum":"1","mixer":"Master","modules":"","script":""},`
?
> volumio@volumio:/volumio$ amixer -c 1
Simple mixer control 'Master',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 248
  Mono: 145 [58%]
Simple mixer control 'Channels',0
  Capabilities: volume
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 248
  Front Left: 194 [78%]
  Front Right: 194 [78%]

volumio, 2016-07-23:
Fixed with this 
https://github.com/volumio/Volumio2/commit/0404fc82eee2e4c6505e17e2b27f085819257b2d

Thanks for reporting!

closed 2016-07-23 by volumio

fanysoft, 2016-09-27:
Hi, same problem here (with hifiberry digi+), when I can expect this fix in my system ?

volumio, 2016-09-27:
https://volumio.org/forum/volumio-final-release-very-near-pre-test-needed-t4709-10.html#p21608

pascal-dev, 2016-10-20:
Hello, same problem with Hifiberry DAC (and Raspberry Pi 1) : one cannot adjust volume trough the Volumio 2 web User Interface.
The HiFiBerry DAC does not have an ALSA mixer control.
Using "alsamixer" through SSH, I get the message: (AlsaMixer v1.0.28) Card: snd_rpi_hifiberry_dac: This sound device does not have any controls. 
It's disappointing because volume adjustment works with Volumio 1.

pascal-dev, 2016-10-24:
I found a work around. (I'm using version 2.001.)
-> Re-select both Output Device and DAC Model manually and Save. 
After that, select {Mixer Type : Software} and {Mixer Control Name : Software Volume Control} (it was not possible before) and Save.
After the first reboot, I got the message "Error : Failed to open audio output" but after a second try, it works as expected !
(Moreover, using "alsamixer" through SSH, one can get the volume level selected in the web UI.)