volumio/Volumio2 #447: apt-get installs on Volumio2 image

macmpi, 2016-05-30:
Hi,
With the new squashed image layout, it seems one can not add components over the std package, to make some custom stuff on top.
Things like [this](https://github.com/volumio/Volumio2/issues/327) are illustrations (even if that particular one may be fixed on next release).
One may want to add some GPIO features which may require to add new libraries & such (for instance [this](http://www.barryhubbard.com/raspberry-pi/howto-raspberry-pi-raspbian-power-on-off-gpio-button/) or whatever).

Would it be possible to have Volumio image built with more of such flexibility so that apt-get installs may work in most cases?
Not re-inventing the wheel, Berryboot may help there: it provides such framework quite nicely (OpenElec/LibreElec have similar approach), and leveraging it may provide other niceties (referring other [issue](https://github.com/volumio/Volumio2/issues/363))

Thanks for consideration

volumio, 2016-05-30:
Sorry but I don't get what you mean...
With this layout you can install pretty much anything... Could you be more specific regarding your issue?

macmpi, 2016-05-30:
Just as an example while I try to install GPIO related hack following [this](http://www.barryhubbard.com/raspberry-pi/howto-raspberry-pi-raspbian-power-on-off-gpio-button/) which requires to install gcc:
`volumio@volumio:~/RPi.GPIO-0.6.2$ sudo apt-get install gcc
Reading package lists... Done
Building dependency tree
Reading state information... Done
You might want to run 'apt-get -f install' to correct these:
The following packages have unmet dependencies:
 gcc : Depends: cpp (>= 4:4.9.2-2) but it is not going to be installed
       Depends: gcc-4.9 (>= 4.9.2-1~) but it is not going to be installed
 libpam-systemd : Depends: systemd (= 215-17+deb8u4) but 215-17+deb8u3 is to be installed
 systemd : Depends: libsystemd0 (= 215-17+deb8u3) but 215-17+deb8u4 is to be installed
E: Unmet dependencies. Try 'apt-get -f install' with no packages (or specify a solution).`

volumio, 2016-05-30:
This depends on raspbian repo and dependencies, and has nothing to do with the squash+overlay layout.  To solve simply do apt-get -f install 

or simply install build-essential

macmpi, 2016-05-30:
had similar issues with several other installs.
apt-get -f install  does not solve the issue, and same problems with build-essential.
I guess the original image may need some update

volumio, 2016-05-30:
Small question:
did you do apt-get update  before trying to install something?
Did you add other sofware repositories?

macmpi, 2016-05-30:
yes I did do apt-get update, and no did not install other repository.
Using stock RC1.
Works on your side?

volumio, 2016-05-30:
I'm flashing a new image to verify that...

volumio, 2016-05-30:
I confirm that in old releases (like yours this was an issue). On newer ones it's isnt.
A new beta is about to be relased ... hold on in the meanwhile.
However you can achieve what you want with a simple bash script... Let me work a bit on that...

macmpi, 2016-05-30:
ok, good to know.
Thanks a lot for checking.

apinner, 2016-06-16:
Closing as this isn't currently an issue, feel free to keep discussing though.

closed 2016-06-16 by apinner