volumio/Volumio2 #452: Volumio 2 on Rasperry Pi 3 WebGUI blank on Android Samsung

BiggerJim, 2016-06-05:
Hiya! Following the 

https://github.com/volumio/Volumio2/issues/266

thread I got Volumio 2 .861 working on my Raspberry Pi 3. Wifi and everything - very cool! 

The only issue is - I can't control it from my phone (Samsung SM-G900F, Android v5), either by Firefox, the native Android browser, or the Volumio Android App. When I enter the ip address, they all just show a completely blank page; whereas I can view the webGUI fine on any Windows or Linux machine on the network.

Any ideas?

Great work, peeps BTW.

BiggerJim, 2016-06-05:
d'Oh! Wasn't logged into WiFi - ie, had no access to local network. Firefox on Android working fine now (though the Volumio App isn't)

apinner, 2016-06-06:
I don't think the android app will work for Volumio 2, it was designed for 1.55.

I am sure it will get updated once 2 is officially released.

closed 2016-06-06 by apinner

BiggerJim, 2016-06-06:
Thanks for the reply; fingers crossed it gets done soon :-)

apinner, 2016-06-07:
You can just use the add to home screen feature that will pretty much present the UI as a native app anyway.

https://developer.chrome.com/multidevice/android/installtohomescreen