volumio/Volumio2 #453: Sound distorted only when 'Volume normalization' is Off

codewarriorFX, 2016-06-06:
The music sounds really distorted only when 'Volume normalization' is Off. Issue happens on Odroid C2 connected to a USB DAC called Serenade Pro DSD/DXD.

apinner, 2016-06-06:
@gkkpch may be best to advise you on this one. I don't think there is any 'official' support for Odroid just yet.

gkkpch, 2016-06-06:
Developers have been testing Volumio 2 features mostly by using a PI.
I have tested as much as possible on 4 different Odroid devices but have not covered this particular option yet. I will try tonight with an xmos usb dac and let you know whether I can reproduce the issue.
It is good to have it on the list, as I don't know whether it was tested enough on a PI. I can do that too for cross reference in case my Odroid C2/ DAC combination fails as well.

As for 'official' support for Odroid: it is our plan to support Odroids from the moment we release Volumio. Next to the PI's and x86 they are the only devices with a complete set of build scripts just now.
Whenever a RC is being released, the Odroid versions have been available too. Sofar they were downloadable from a HK developer's website. With RC2 this should actually change so we can test the update procedure. Michelangelo and I will be working on that.

codewarriorFX, 2016-06-06:
> I will try tonight with an xmos usb dac and let you know whether I can reproduce the issue.

@gkkpch Hi, just for the reference when doing tests, the USB DAC I use (Serenade Pro) it has the C-Media CM6632A usb-to-i2s processor and the dac used is the burr-brown pcm1795.

So the high possibility to replicate the issue is when using a USB DAC with this C-Media CM6632A audio processor.

gkkpch, 2016-06-06:
yeah, I have a whole stack of dac's available to test :) Not.....
Anyway, there are definitely 2 things wrong with the playback setting menu
1. it does not store the data properly and once changes, you can't undo it in any way. The mpd config and volumio configuration data seem out-of-sync
2. changing to Volume Normalisation OFF stopped my playback completely and I could not get it back to working (yet)

codewarriorFX, 2016-06-27:
@gkkpch Testing now with RC2, and sounds way better. Also noticed when reading hw parameters (cat /proc/asound/card2/pcm0p/sub0/hw_params), Is getting now at the bit depth it supposed to be, not 16-bit,

access: RW_INTERLEAVED
format: S24_3LE
subformat: STD
channels: 2
rate: 96000 (96000/1)
period_size: 12000
buffer_size: 48000

gkkpch, 2016-06-28:
OK, good to hear. Please close the issue when you believe it is solved.

codewarriorFX, 2016-06-28:
Hi, I tested volume normalization OFF with FLAC, AIFF and DSD (64 & 128) and no problems. but unfortunately still distorts on MP3 files. When I get home after work, I will try disabling the odroid dac drivers since Im using a USB dac and try again.

BTW, Spotify works awesome!

gkkpch, 2016-06-28:
Disabling the odroid dac drivers on RC2 should not be needed unless you have a hifi shield or another DAC connected to the i2s bus. Even then it would not make any difference as your usb dac is using the standard linux usb audio sound driver. If you still want to remove the odroid dac drivers, edit ` /etc/modules` and comment the two driver lines.

gkkpch, 2016-06-28:
Have you tried rebooting the C2 first after you switched to the USB DAC, and did it still distort on MP3s  then? I verified the last 10 minutes on my C2 with an XMOS (pcm5101) USB DAC, it plays anything, MP3, FLAC, OGG, APE, WAV upto 352.8Khz, 16bit, 24 bit (did not try 32bit/384Khz). I do not notice any distortion, strange.

codewarriorFX, 2016-06-28:
It still distort after reboot. The issue may be pointing to the C-Media CM6632A (with PCM1795) that my USB dac uses. I will try using HDMI to my A/V receiver if still do the same. Sadly, I don't have a XMOS-based USB dac to test at this moment.

By the way, I tested a sample AIFF file at 32-bit/384kHz with no issue.

gkkpch, 2016-06-28:
Ah, nice to know AIFF works...

volumio, 2016-06-28:
We found out that volume normalization resamples to 24 bits. This might be a clue.

Plus, CM6632 has a notorious bug with alsa: when bitrate\bitdepth changes (like playing a 44.1/16 then a 24/44.1) it gets confused and sticks to old resolution. A dirty fix is usually to stop and restart playback...

codewarriorFX, 2016-06-28:
Ok, here is the results of the Odroid C2 connected to the Yamaha RX-A800 receiver (HDMI). Please note that this receiver goes up to 24/192 and PCM only. DSD files wont play since the DAC does not support it.

-MP3 16/44.1 - OK
-Spotify(16/44.1) - OK
-FLAC 16/44.1 - OK
-FLAC 24/88.2 - OK
-FLAC 24/96 - OK
-FLAC 24/192 - No Sound
-Any file higher that 24-bit and/or 192 - No Sound

I noticed when switching to 24/192 on HDMI wont give sound. It maybe C2 hardware issues.

volumio, 2016-06-28:
Something related to C2 HDMI Kernel driver

codewarriorFX, 2016-06-29:
From the comment about the CM6632, is this dirty fix can be done from the terminal (SSH)?

volumio, 2016-06-29:
Yep. play a song then

mpc stop
mpc play

closed 2017-06-02 by volumio