volumio/Volumio2 #461: [question] Get string values from...

codewarriorFX, 2016-06-15:
Hello - Is there any way to get the string values from the song currently playing on Volumio 2? My plan is to get these string values to show it on a oled display connected to my Odroid C2 board. Im looking for: song name, artist name, album name, resolution , format and current playing time. Please let me know if this is possible.

Thanks

volumio, 2016-06-20:
Yes, you can use a websocket client (like for example a little node script)  and listen for 'getState' event...

This is an example: 

var io=require('socket.io-client');

var socket= io.connect('http://localhost:3000');

socket.emit('getState', '');

socket.on('pushState ',function(data)
{
    console.log(JSON.stringify(data));
});

codewarriorFX, 2016-06-20:
Do I always have to close the connection after getting the info?

volumio, 2016-06-20:
If you don't close it, you'll get notifications when the song changes etc... So for your purpose you may want to keep it open..

codewarriorFX, 2016-06-20:
Does these notification is on the same output?

volumio, 2016-06-20:
Yep, show me your code and I'll help you if you need

codewarriorFX, 2016-06-21:
Im using the code below, and is based from this: https://volumio.org/forum/lcd-track-information-problem-t1689.html.

from mpd import MPDClient
from oled.device import ssd1306, sh1106
from oled.render import canvas
from PIL import ImageFont
from PIL import Image
import time
import threading
import logging

...........

def mpd_info(lock):
	print "mpd_info start"
	global song_display
	global album_display
	global artist_display
	
	song_display = " "
	album_display = " "
	artist_display = " "

	song_name = " "
	album_name = " "
	artist_name = " "
	
	client = MPDClient()               # create client object
	client.timeout = 10                # network timeout in seconds (floats allowed), default: None
	client.idletimeout = None          # timeout for fetching the result of the idle command is handled seperately, default:$
	client.connect("localhost", 6600)  # connect to localhost:6600
	
	while True:
		currentsong = client.currentsong()

		print ""
		#print 'CURRENT----------------------------------------------------------------'
		if len(currentsong) > 0:
			for text in currentsong:
				if text == "album":
					album_name = str(currentsong.get(text))

				if text == "artist":
					artist_name = str(currentsong.get(text))

				if text == "title":
					song_name = str(currentsong.get(text))

				#print text + ": " + str(currentsong.get(text))

			current_id = int(currentsong.get("pos")) + 1
		else:
			print "No current song (empty playlist)"
		
		if song_display != song_name:
			lock.acquire()
			try:
				logging.debug('Locking')
				song_display = song_name
				album_display = album_name
				artist_display = artist_name
				#time.sleep(1)
			finally:
				logging.debug('Not locking')
				lock.release()

		time.sleep(1)
		
	client.close()                     # send the close command
	client.disconnect()                # disconnect from the server

.................

if __name__ == "__main__":

	print "Main started"
	
	lock = threading.Lock()
	
	mpd_info = threading.Thread(target=mpd_info, args=(lock,), name='MPD Info')
	mpd_info.setDaemon(True)
	mpd_info.start()

	draw_oled = threading.Thread(target=draw_oled, args=(lock,), name='Draw OLED')
	draw_oled.start()

codewarriorFX, 2016-06-21:
This is one of my first codes written in Python, and with the examples about threading, locks, functions found on google I was able to get this working (mostly...) Please bear with me =)

If you noticed on the if statement song_display != song_name, Im locking the other thread that draws the information on the OLED. When I change to another song, the oled thread cannot read the values until those variables are updated, and then release again.

Some issues Im facing is when during the event of changing song, at some point the mpd function hangs and it does not update the info on the display. It does not give error or crash. If I change to another song again, then it updates the info and continues normally. 

And sometimes, it delays around 6-10 seconds to show the new info.

Any suggestions?

volumio, 2016-06-21:
Sorry but I don't really know Python very well..

Idea: why don't we do somethign with node? I saw this module:
https://github.com/fivdi/lcd

Which just seems perfect. Plus if we write it in node, this can become a plugin...
What do you think?

codewarriorFX, 2016-06-21:
Oh man, I wish but I really don't know about Node :( Maybe someone here with that knowledge can grab a Odroid C2 and adapt the Hitachi HD44780 code.

codewarriorFX, 2016-06-21:
Look at the progress made. It needs many improvements,

https://www.youtube.com/watch?v=SbO2cSzbsU4

codewarriorFX, 2016-06-23:
Hello,

I have resolved the timing issues between the threads by using the MPD Client idle command. No delays or hangs when showing the info on the OLED display. But, it will needs some refinements since it crashes when I change the Output devices. Maybe I can try adding a 'retry MPD connection' function on the script when this happens.

BTW, when RC2 will be available for the Odroid C2? I can't wait to test spotify and airplay with this script.

apinner, 2016-06-27:
RC2 for Odroid is available now i think.

Going to close this as it's not an ongoing issue.

closed 2016-06-27 by apinner