volumio/Volumio2 #462: Support SFTP

clsidxxl, 2016-06-15:
I like manage the Volumio files with MobaXterm SFTP session,please support SFTP in Volumio 2

apinner, 2016-06-16:
Which files do you want to manage?

clsidxxl, 2016-06-16:
It's more easy and friendly to modify  for example the systeme file of DLNA (files.db,art cache),and other files ex.boot.ini or dtb, easily add credential file,modify Fstab,minidlna.conf,mpd.conf......

apinner, 2016-06-16:
You can do all of this using SSH

clsidxxl, 2016-06-16:
MobaXterm connect in SSH  to Odroid and  it's more easy manage file with SFTP

clsidxxl, 2016-06-16:
you know MobaXterm SFTP?

apinner, 2016-06-16:
Nope, never heard of it

clsidxxl, 2016-06-16:
[http://www.zimagez.com/zimage/sanstitre92f51760d2aadaa4dcd9bfdb942d9eba.php](url)

volumio, 2016-06-16:
Volumio has sssh/sito enabled by default.  Doesn't work with mobaxtem?

clsidxxl, 2016-06-16:
no,don't work; tomorrow I will again install Volumio 2 and try again,I give you the error message when I want to copy a file in SFTP  session.

xipmix, 2017-01-30:
HI, is this still a problem? If not please could you close this again.

closed 2017-06-02 by volumio