volumio/Volumio2 #465: calculate max. output volume in frontend relative to max. output volume of system settings

HolyMacarony, 2016-06-17:
the problem: 
we set the maximum output volume of volumio to say 60% in the system settings, because we'd like to prevent the input signal for the stereo from being to strong. 
This leads to a cropped volume bar in the user frontend, which makes it harder to set the volume with a good granularity and wastes the space of the volume bar, which can't be used to alter the volume.

I'd like to suggest the following solution:
If the user sets the maximum output volume to let's say 60% in the system backend. Then it would be great if the frontend volume input of the user is calculated relative to the customized max. output volume of the backend. 
Meaning 100% in frontend GUI represents the 60% max. Output for my stereo which i defined previously in the system settings.
So the granularity of setting the volume as user is much better because the whole volume bar can be used instead of just the cropped bottom end of it. 

What do you think? Thank you very much for this great piece of software!

apinner, 2016-06-17:
Its a nice idea, the only issue i foresee is people will think that setting the max volume hasn't worked if somehow its not indicated in the UI.

One solution could be to change the scale of the control knob so its goes from 0 to n rather than 100. The other option would be to indicate that max volume is n% somewhere in the middle of the knob and keep the scale at 0-100.

HolyMacarony, 2016-06-17:
you are right, it is currently established, that a lowered max. output volume is indicated by a "cropped" volume bar.

So in my opinion, setting the top 100 to the predefined max. output value n or n% would be sufficient and self explanatory.