volumio/Volumio2 #475: Plugin spotify-connect-web

frequencx, 2016-06-24:
I currently use Fornoths spotify-connect-web (https://github.com/Fornoth/spotify-connect-web) which allows me to use my Volumio1 instance as a Spotify Connect Device. This is great because I can use the Spotify UI as usual and switch to my amp seamlessly. 

With an integration in Volumio2 I was hoping to achieve proper switching between Airplay and Spotify connect.

Can anybody offer some guidance on how to implement spotify-connect-web as a plugin for Volumio2? 
Can it be done? Will it be done? Has it been done?

Thanks!

mundodisco8, 2017-04-03:
Also, multi-account support would be killer!

exetico, 2017-04-09:
@mundodisco8  - Take a look at [this](https://github.com/balbuze/volumio-plugins/blob/master/plugins/music_service/volspotconnect2/volspotconnect2.zip).

mundodisco8, 2017-04-10:
As a matter of fact, I tried it this weekend, and it works exactly as I expect.

I've been researching a bit, and apparently spotify-connect-web uses libspotify, which is a library that Spotify, apparently, has no intention to continue developing (I asked for a key a week ago and they haven't even replied). So I found about librespot, which seems to be a reverse engineered, open source alternative to libspotify, and also about Balbuze's plugin, and it works like a charm. It even does more than I expected, by allowing anyone with a premium account connected to my network to stream to my volumio. What I originally meant was that I wanted to connect to it through two different premium accounts: this has a mode where you can do that, or allow anyone with a with a premium account in your network to connect. This could be an issue in, for example, your work's network, but not a problem at home.

So, I would say that this issue can be closed, as Balbuze has two plugins that do the trick, one based on librespot, and one based around libspotify, if for any reason still need to use that.

volumio, 2018-03-26:
Pkugin available

closed 2018-03-26 by volumio