volumio/Volumio2 #482: Volumio RC2: after installing and using Spotify plugin, library and webradio won't work

gkkpch, 2016-06-26:
This is a persistent issue on the Odroid C2, reboot does not have any influence.
Issue needs further investigation, just to make sure, I will also try it on a PI 3.

gkkpch, 2016-06-26:
Tried it on a PI, no problem
Seems Odroid-related, on a C2 it will continue to play Spotify, but anything else keeps failing.
Checked mpd.conf, shows hw:1,0 correctly. Startup sound is played correctly as well.
I will re-flash and try the same again..

gkkpch, 2016-06-26:
Same issue after re-flashing. But not sure, it looks more like the NAS connection gets lost and library is not playing anymore for that reason. After re-editing the NAS connection with proper uid/pwd seems to resolve it. So, it seems the NAS connection does not get re-connected properly upon restart, likely no relation with Spotify at all. So, issue 1 perhaps pinpointed. 
Not playing webradio will need another re-flash, will report....

gkkpch, 2016-06-26:
easy, webradio does not do anything.
So I'm rephrasing my initial issue: 
- Using a NAS with uid/pwd on a cifs account can lead to NAS not being re-connected after rebooting.
- Webradio doesn't not work at all (tried by country, dirble, Netherlands)

SonosKiller, 2016-06-26:
I have a similar problem 

Running a Rpi3 a clean install of the RC2 on a new Sd card using no Dac just the audio jack.

Mp3 music works fine!
But no webradio channels works. The only sound I hear is like the startup sound after a few seconds after I started the channel 

I can't change the volume in the playback dialog 
note: the "Mixer control" in "Playback options"is empty and not changeable

Tried the sd card in an Rpi2B and done a re-flash  same problem

gkkpch, 2016-06-26:
@sSonosKiller: please open another issue for that, my issue is NAS and uid/pwd related as it seems!!

gkkpch, 2016-06-26:
I wonder, if this only happens the first time after a NAS connection with uid/pwd is configured as after several restarts, there seems to be no problem...

gkkpch, 2016-06-26:
So, after testing two more re-flashes, it seems the uid/ pwd goes lost after the first time configuration.
When corrected later, the configurations seems to stay OK.

klokslagB, 2016-06-26:
To add to the webradio problem. After adding an url stream at first I could play. Don't  know when I lost it but couldn't restart the same mp3 url next try. I also noticed that there wasn't any info like artist displaying which it did in RC1. Everything else looking fine. keep up the good work!

gkkpch, 2016-06-26:
no use adding comments to an issue that has no direct relation to your webradio. Please open a separate issue for the web radio issue, much appreciated...

SonosKiller, 2016-06-26:
Okey ! 
Done !

closed 2017-06-02 by volumio