volumio/Volumio2 #484: Volumio RC2: No Webradio and can't adjust volume

SonosKiller, 2016-06-26:
Running a Rpi3 a clean install of the RC2 on a new Sd card using no Dac just the audio jack.

Mp3 music works fine!
But no webradio channels works. 
The only sound I hear is the startup sound after a few seconds 

I can't change the volume in the playback dialog
note: the "Mixer control" in "Playback options" is empty and not changeable

Tried the sd card in an Rpi2B and done a re-flash, same problem

apinner, 2016-06-27:
I can confirm that i am also having an issue with shoutcast radio stations.

Nothing happens, i just get a shoutcast URL show in the playback information.

apinner, 2016-06-27:
[image]

SonosKiller, 2016-11-29:
Old issue ! fixed in current version 2.031

closed 2016-11-29 by SonosKiller