volumio/Volumio2 #489: RC2: can not modify/suppress MyWebRadio

macmpi, 2016-06-27:
Hi,

Adding webradios in the list works (though won't playback, seems a general webradio issue at that point), but once added, they can not be edited / suppressed.

nosferatujr, 2016-06-30:
I have similar issue. Added webradios can't  be played.
the generated file /data/favourites/my-web-radio looks good:
[
  {
    "service": "webradio",
    "name": "RAC1",
    "uri": "http://www.rac1.org/audio/Player/rac1.pls"
  }
]
BTW macmpi you can delete by edit this file. keep the 
[
]
or you will get a browser error

skikirkwood, 2016-07-01:
Running on RC2 Hotfix, you can now delete/edit your My Web Radios stations.  None of them will play though.  Tried several, with Soma FM Groove Salad being the latest.  File looks good though:

volumio@volumio2:/data/favourites$ more my-web-radio 
[
  {
    "service": "webradio",
    "name": "Soma FM Groove Salad",
    "uri": "http://ice1.somafm.com/groovesalad-128-aac"
  }
]

nosferatujr, 2016-07-02:
Thanks, Now is working :-)

HarrisonIto, 2016-07-02:
Added webradios still can't be played on my RC2.
I tried to add and play 'Soma FM Groove Salad' which skikirkwood showed, but sitll can't.

volumio@volumio:/data/favourites$ cat my-web-radio
[
  {
    "service": "webradio",
    "name": "FMK",
    "uri": "http://hdv4.nkansai.tv/kawaguchi"
  },
  {
    "service": "webradio",
    "name": "FMK e",
    "uri": "http://radio.elsatower32.org:8080/stream/1/"
  },
  {
    "service": "webradio",
    "name": "FMK d",
    "uri": "http://hdv4.nkansai.tv/kawaguchi?MSWMExt=.asf"
  },
  {
    "service": "webradio",
    "name": "test",
    "uri": "http://ice1.somafm.com/groovesalad-128-aac"
  }
]

klokslagB, 2016-07-11:
I don't know if this a separate issue but I can't play my mp3 radio's. The shoutcast radios I can play without problem. But not my added Belgian radio streams. Also using Dirble the same radio doesn't play.

Example: http://mp3.streampower.be/stubru-high.mp3

on RC2 I could play not a single radio.
Currently on RC2 hotfix + dev only shoutcast. 
RC1 played without problem.

should be easily reproducible

macmpi, 2017-01-11:
cleanup: original issue fixed a while back.

closed 2017-01-11 by macmpi