volumio/Volumio2 #490: RC2: Failed NAS mounts for path with <space> character

macmpi, 2016-06-27:
Hi,

Mounting a NAS path containing <space> character fails, although that same path is recursively parsed fine (and music played) if mount is set at a parent folder.
Would be nice to remove such limitation (tested under NFS).
Thanks.

volumio, 2016-06-27:
Thanks for reporting

macmpi, 2016-06-27:
and thanks for all the hard & amazing work!
(too bad the issues thread only enlighten problems, whereas this RC2 is a great overall achievement!)

apinner, 2016-06-27:
Pretty sure a fix has been pushed for this; https://github.com/volumio/Volumio2/commit/2afa847c546bd7874c485c0f33777f09cf8fdbd2

closed 2016-06-27 by apinner

clemensmartin, 2016-06-28:
While you are fixing this issue, it would be a good idea to keep the data fields filled when after an error occurs. 
In the current (RC2) version, all fields are set to blank if the mount does not succeed. 
Further, with an unsuccessful mount I end up with a bunch of mount points in the file system, but these don´t show in the user interface. A bit of feedback on the success or failure of a mount would be helpful.

apinner, 2016-06-28:
@clemensmartin That is a separate issue, can you please create a new one.

clemensmartin, 2016-06-28:
will do
C
#