volumio/Volumio2 #496: Volumio2 RC2 for Odriod C2 - no acces from browser

johnyn6, 2016-06-29:
I can´t connect from my browser. I use DHCP, router give IP, but only ssh, scp is working (putty, winscp). Ipconfig / flushdsn doesn´t help.
With RC1 I have no problem.
My config is C2 + 8GB emmc and ethernet cable, no wifi.

volumio, 2016-06-29:
Could you add :3000 to your IP and see if that works?

johnyn6, 2016-06-29:
Port 3000 doesn´t work.

volumio, 2016-06-29:
If  you can ssh..
sudo journalctl -p 7 > /home/volumio/faillog.log

And paste the content of faillog.log here, if that fits (or try to get the error part)

johnyn6, 2016-06-29:
[faillog.zip](https://github.com/volumio/Volumio2/files/339971/faillog.zip)

volumio, 2016-06-29:
The mdns stack is failing, I don't know why.. 
Do you have multiple volumios? 
Which router do you have? Does it have a firewall?

mardev7, 2016-06-29:
(Volumio2 RC2, Odroid C2 Hifi Shield)
It works connecting from my browser using http://volumio.local when I am connecting with a cable to my router. However, if I connect to my router using wifi (using my phone) it fails. Then I must use the ip number to be able to connect.

apinner, 2016-06-29:
Are you using an Android phone by any chance?

johnyn6, 2016-06-29:
I can test RC2 on C1+. RC1 is working on both odroids. I have router with openwrt and yes, it have firewall.

mardev7, 2016-06-29:
@johnyn6 Yes, Android on my phone (and dd-wrt on my router).

apinner, 2016-06-29:
I think that may be your problem. http://volumio.local won't work on android, you have to use the IP address. Its because android doesn't support mDNS, nothing that we can do to fix that.

However, if your volumio always has the same IP through either static or a dhcp reservation then you can use the add to homescreen feature in your mobile browser and the Volumio UI will open fullscreen like a native app.

mardev7, 2016-06-29:
@apinner  Ok, great, thanks for the info! 
I just got my odroid and I am really impressed with Volumio, once the volume control is implemented it will be perfect!

johnyn6, 2016-06-30:
On C1+ RC2 is working... So there is some problem in C2 build.

johnyn6, 2016-08-01:
With newest RC2Fix-2016-07-31 still no web access. I want to try my C2 with another router, or only with fixed IP directly to my PC.

closed 2017-06-02 by volumio