volumio/Volumio2 #499: [question] Check Airplay status...

codewarriorFX, 2016-06-30:
Hi - I would like to know if there is a similar way for Airplay to check via telnet if is playing, pause or stopped, just like MPD and spop?

I have plans in mind to create a script that if I have a device connected to my Odroid via Airplay, show the Airplay logo on the oled screen.

codewarriorFX, 2016-06-30:
This is on Odroid C2 with RC2.

paviro, 2016-07-11:
Would also be nice if the webUI would show the status! Especially since [shairport-sync](https://github.com/mikebrady/shairport-sync) seems to be able to transfer metadata!

volumio, 2016-07-11:
Yes, I started doing that (look at the Airplay_WIP branch), but it's not as simple as it seems...

paviro, 2016-07-11:
Mhh really does not look that simple :D So chances are it is going to be there one day?

volumio, 2016-07-11:
I hope so...

codewarriorFX, 2016-07-11:
I tried to add the flag "-M <directoryname>" on the shairport command, but it doesn't seem to show the now_Playing file on the directory assigned.

Nope, is not that simple as it seems...

closed 2017-06-02 by volumio