volumio/Volumio2 #504: [RC2] Can't connect after system update

pmd2016, 2016-07-01:
Hi!

I just updated RC2 through the web interface. After restart, I can't connect to volumio through the web interface. I can login through ssh.

I have no idea where to look for a solution. Any ideas?

Cheers,
Michiel

apinner, 2016-07-01:
Have you tried turning it off and on again?

pmd2016, 2016-07-02:
I did a reboot via SSH.

pmd2016, 2016-07-02:
This is the output of the log file. I just noticed the date is off...

`1970-01-01T00:00:25.936Z - info: -------------------------------------------
1970-01-01T00:00:25.949Z - info: -----            Volumio2              ----
1970-01-01T00:00:25.951Z - info: -------------------------------------------
1970-01-01T00:00:25.952Z - info: -----          System startup          ----
1970-01-01T00:00:25.952Z - info: -------------------------------------------
1970-01-01T00:00:26.397Z - info: Plugin folders cleanup
1970-01-01T00:00:26.399Z - info: Scanning into folder /volumio/app/plugins/
1970-01-01T00:00:26.400Z - info: Scanning category audio_interface
1970-01-01T00:00:26.403Z - info: Scanning category audio_interfaces
1970-01-01T00:00:26.442Z - info: Scanning category miscellanea
1970-01-01T00:00:26.444Z - info: Scanning category music_service
1970-01-01T00:00:26.446Z - info: Scanning category music_services
1970-01-01T00:00:26.459Z - info: Scanning category plugins.json
1970-01-01T00:00:26.461Z - info: Scanning category system_controller
1970-01-01T00:00:26.464Z - info: Scanning category user_interface
1970-01-01T00:00:26.466Z - info: Scanning into folder /data/plugins/
1970-01-01T00:00:26.467Z - info: Plugin folders cleanup completed
1970-01-01T00:00:26.471Z - info: Loading plugins from folder /volumio/app/plugins/
1970-01-01T00:00:26.486Z - info: Loading plugins from folder /data/plugins/
1970-01-01T00:00:26.493Z - info: Loading plugin "system"...
1970-01-01T00:00:26.589Z - info: Loading plugin "appearance"...
1970-01-01T00:00:27.334Z - info: Loading plugin "i2s_dacs"...
1970-01-01T00:00:27.357Z - info: Loading plugin "network"...
1970-01-01T00:00:27.491Z - info: Loading plugin "services"...
1970-01-01T00:00:27.501Z - info: Loading plugin "alsa_controller"...
1970-01-01T00:00:27.685Z - info: Loading plugin "volumio_command_line_client"...
1970-01-01T00:00:27.706Z - info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
1970-01-01T00:00:27.709Z - info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
1970-01-01T00:00:27.710Z - info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
1970-01-01T00:00:27.767Z - info: Loading plugin "volumiodiscovery"...
1970-01-01T00:00:27.900Z - info: Applying required configuration parameters for plugin volumiodiscovery
1970-01-01T00:00:27.990Z - info: CoreCommandRouter::executeOnPlugin: system , registerCallback
1970-01-01T00:00:27.993Z - info: Loading plugin "upnp"...
1970-01-01T00:00:28.007Z - info: [28006] Starting Upmpd Daemon
1970-01-01T00:00:28.037Z - info: CoreCommandRouter::executeOnPlugin: system , registerCallback
1970-01-01T00:00:28.039Z - info: Loading plugin "airplay_emulation"...
1970-01-01T00:00:28.050Z - info: [28049] Starting Shairport Sync
1970-01-01T00:00:28.056Z - info: Loading plugin "mpd"...
1970-01-01T00:00:28.170Z - info: Loading plugin "networkfs"...
1970-01-01T00:00:28.211Z - info: Loading plugin "alarm-clock"...
1970-01-01T00:00:28.457Z - info: Loading plugin "webradio"...
1970-01-01T00:00:29.823Z - info: Loading plugin "albumart"...
1970-01-01T00:00:29.856Z - info: Plugin example_plugin is not enabled
1970-01-01T00:00:29.858Z - info: Loading plugin "updater_comm"...
1970-01-01T00:00:30.263Z - info: Plugin mpdemulation is not enabled
1970-01-01T00:00:30.264Z - info: Loading plugin "websocket"...
1970-01-01T00:00:30.327Z - info: ___________ START PLUGINS ___________
1970-01-01T00:00:30.338Z - info: CoreCommandRouter::volumioAddToBrowseSources[object Object]
1970-01-01T00:00:30.340Z - info: [30339] CoreMusicLibrary::Adding element Webradio
1970-01-01T00:00:30.361Z - info: Loading i18n strings for locale en
1970-01-01T00:00:30.449Z - info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
1970-01-01T00:00:30.450Z - info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
1970-01-01T00:00:30.452Z - info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
1970-01-01T00:00:30.453Z - info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
1970-01-01T00:00:30.454Z - info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
1970-01-01T00:00:30.476Z - info: BOOT COMPLETED
1970-01-01T00:00:30.551Z - info: [30549] CoreStateMachine::resetVolumioState
1970-01-01T00:00:30.553Z - info: [30552] CoreStateMachine::getcurrentVolume
1970-01-01T00:00:30.555Z - info: CoreCommandRouter::volumioRetrievevolume
1970-01-01T00:00:30.588Z - info: [30585] CoreStateMachine::updateTrackBlock
1970-01-01T00:00:30.594Z - info: [30590] CorePlayQueue::getTrackBlock
1970-01-01T00:00:30.681Z - info: Cannot read play queue form file
1970-01-01T00:00:30.912Z - info: MPD running with PID456 ,establishing connection
1970-01-01T00:00:30.954Z - info: Setting Device type: Raspberry PI 2/3
1970-01-01T00:00:31.001Z - info: [31000] VolumeController:: Volume=undefined Mute =undefined
1970-01-01T00:00:31.007Z - info: [31006] CoreStateMachine::pushState
1970-01-01T00:00:31.010Z - info: [31009] CoreStateMachine::getState
1970-01-01T00:00:31.012Z - info: [31011] CorePlayQueue::getTrack 0
[... 1 lines cut ...]
1970-01-01T00:00:31.017Z - info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
1970-01-01T00:00:31.023Z - info: [31022] InterfaceWebUI::pushState
1970-01-01T00:00:31.242Z - info: Pushing Favourites {"service":"mpd","uri":"","favourite":false}
1970-01-01T00:00:31.406Z - info: Setting volume on startup at disabled
[... cut ...]
1970-01-01T00:00:31.566Z - info: Pushing Favourites {"service":"mpd","uri":"","favourite":false}
1970-01-01T00:00:31.602Z - info: mDNS: Found device Volumio
1970-01-01T00:00:31.833Z - info: [31831] Shairport-Sync Started
1970-01-01T00:00:31.852Z - info: Upmpdcli Daemon Started
2016-07-01T18:38:24.749Z - info: mDNS: A device disapperared from network
2016-07-01T18:38:24.957Z - info: mDNS: Found device Volumio
2016-07-01T18:38:59.448Z - info: mDNS: A device disapperared from network
2016-07-01T18:39:00.564Z - info: mDNS: Found device Volumio`

xarath, 2016-07-05:
Same here.  I've set the date correctly then rebooted but it still doesn't work.

sennahLessert, 2016-07-10:
i made the same experiences.
My device is a Rasperry Pi 2 with hifiberry dac+

After that i did a

> 
rm -rf /volumio
cd /
git clone https://github.com/volumio/Volumio2.git volumio
cd volumio
npm install
> 

compiled everything and performed a reboot.
Again no web-interface. After that i did
>
node index.js
>

that worked well but still no webinterface , but http://ip:3000/dev was accessible
Any idea what to do?

volumio, 2016-07-10:
You guys forgot to add the UI (which is in a different repo).

git clone --depth 1 -b dist --single-branch https://github.com/volumio/Volumio2-UI.git /volumio/http/www

closed 2016-11-20 by volumio