volumio/Volumio2 #507: Spotify playlists not always displayed

skikirkwood, 2016-07-01:
I downloaded and activated the Spotify plugin for RC2 hotfix (June 30 release) and after plugging in my Spotify credentials tried to browse my Spotify playlists.  I had a screen with a folder icon, "Playlists" to the right, and the hamburger icon over on the far right.

Clicking on Playlists didn't result in anything.  I then plugged in my Spotify credentials a couple more times, got the same behavior, and then after a 5th or 6th try it worked.  I was able to play Spotify tracks, see them in the queue, etc.  I then went back to browse more Spotify playlists and same behavior as before.  After clicking around the UI, the playlists appeared again, then they disappeared again!

The Volumio log file shows this when I cannot browse Spotify:

2016-07-01T23:01:20.433Z - info: CoreCommandRouter::executeOnPlugin: spop , handleBrowseUri
2016-07-01T23:01:21.536Z - info: CoreCommandRouter::executeOnPlugin: spop , handleBrowseUri
2016-07-01T23:01:21.536Z - info: [1467414081536] ControllerSpop::sendSpopCommand
2016-07-01T23:01:21.538Z - info: ADDING DEFER FOR COMMAND ls
2016-07-01T23:01:21.547Z - info: FIRST BRANCH
2016-07-01T23:01:21.548Z - info: BEFORE: SPOP HAS 1 PROMISE IN STACK
2016-07-01T23:01:21.548Z - info: AFTER: SPOP HAS 0 PROMISE IN STACK
2016-07-01T23:01:21.551Z - info: FIRST BRANCH
2016-07-01T23:01:21.552Z - info: BEFORE: SPOP HAS 0 PROMISE IN STACK
2016-07-01T23:01:21.552Z - info: AFTER: SPOP HAS 0 PROMISE IN STACK

When the browse command does work I see a list of the Spotify playlists in the log file.  Using the latest versions of Chrome and Safari on a Mac running the latest release of El Capitan.  Overall I'd say 90% of the time I do not get a list of playlists, seems to be random when they do show up.

apinner, 2016-07-02:
Do you have a Spotify premium account?

skikirkwood, 2016-07-02:
Yes, I have a Spotify Premium account and it works fine on Volumio 1.    I could PM my credentials to someone for debugging if that would help.   
  

  
I was able to play Spotify tracks from my playlist, but 90% of the time clicking on the Playlist link resulted in nothing.

skikirkwood, 2016-07-02:
Here's some additional info.  I created a new Spotify Premium account, added a few playlists, and put the credentials of the new account (user: volumio2) into Volumio.   Browsing, playing Spotify playlists worked perfectly.  I then put the credentials of my original (user: altamontstudios) account back into the Spotify plugin in Volumio.  Back to the same issue of browsing Playlists stopped working.  Both are Spotify Premium, one works perfectly, one mostly does not work, but occasionally does.  Strange!

One guess here is I have a very large number of playlists with my altamontstudios account.  Perhaps some kind of timeout issue or something else regarding how large the data there is?

Again, be happy to PM credentials of both accounts if someone can help debug this, although if you are outside the U.S. you would need to VPN into a U.S. server since they are U.S. based Spotify accounts.

skikirkwood, 2016-07-02:
After rebooting my Pi, I no longer see Spotify as an option in Browse.  The plugin is active, with the credentials of my original Spotify account.  I then put the credentials of my new Spotify account in to the plugin and Spotify showed up again under Browse.  Seems like this plugin does not like my Spotify account!  Here's the log after bootup when Spotify would not start:

1970-01-01T00:00:26.018Z - info: [26018] Shairport-Sync Started
1970-01-01T00:00:27.205Z - info: 
[27204] ---------------------------- Client requests Volume 22
1970-01-01T00:00:27.207Z - info: [27206] VolumeController::SetAlsaVolume22
1970-01-01T00:00:27.219Z - info: [27218] ------------------------------ 14ms
1970-01-01T00:00:27.230Z - info: [27229] VolumeController::Volume 22
1970-01-01T00:00:27.231Z - info: [27230] CoreStateMachine::pushState
1970-01-01T00:00:27.231Z - info: [27231] CoreStateMachine::getState
1970-01-01T00:00:27.232Z - info: [27231] CorePlayQueue::getTrack 0
1970-01-01T00:00:27.232Z - info: CoreCommandRouter::volumioPushState
1970-01-01T00:00:27.233Z - info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
1970-01-01T00:00:27.234Z - info: [27234] InterfaceWebUI::pushState
1970-01-01T00:00:27.242Z - info: Pushing Favourites {"service":"webradio","uri":"http://yp.shoutcast.com/sbin/tunein-station.m3u?id=899590","favourite":false}
1970-01-01T00:00:30.345Z - info: Connecting to daemon
1970-01-01T00:00:30.366Z - info: SPOP command error:
1970-01-01T00:00:30.370Z - info:  Error: connect ECONNREFUSED 127.0.0.1:6602
    at Object.exports._errnoException (util.js:1007:11)
    at exports._exceptionWithHostPort (util.js:1030:20)
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1080:14)
1970-01-01T00:00:30.373Z - info: SPOP status error:
1970-01-01T00:00:30.374Z - info:  Error: connect ECONNREFUSED 127.0.0.1:6602
    at Object.exports._errnoException (util.js:1007:11)
    at exports._exceptionWithHostPort (util.js:1030:20)
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1080:14)
2016-07-02T17:08:32.636Z - info: Volumio called home
2016-07-02T17:08:39.386Z - info: mDNS: A device disapperared from network
2016-07-02T17:08:39.489Z - info: mDNS: Found device Volumio2
2016-07-02T17:08:50.768Z - info: mDNS: Found device Volumio2

skikirkwood, 2016-07-04:
Ok, here's one more data point.  I telnet'd to the spop daemon and gave a "ls" command, just as the Volumio 2 Spotify plugin does.  I got back an array of my playlists, but the first entry was an empty object, so perhaps that is the cause of the problem with this one account:

ls
{
  "playlists" : [
    {
    },
    {
      "type" : "playlist",
      "name" : "Smooth Morning",
      "tracks" : 69,
      "offline" : false,
      "index" : 1
    },
    {
      "type" : "playlist",
      "name" : "Alexander Volosnikov – All These Melodies",
      "tracks" : 13,
      "offline" : false,
      "index" : 2
    },
    {
      "type" : "playlist",
      "name" : "Sunny Terrace – The Orbit",
      "tracks" : 10,
      "offline" : false,
      "index" : 3
    },
    {
      "type" : "playlist",
      "name" : "Phoria – Volition",
      "tracks" : 11,
      "offline" : false,
      "index" : 4
    },
    {
      "type" : "playlist",
      "name" : "David Arkenstone – Chillout Lounge",
      "tracks" : 10,
      "offline" : false,
      "index" : 5
    },
    {
      "name" : "Coastsider",
      "type" : "folder",
      "playlists" : [
        {
          "type" : "playlist",
          "name" : "Coastsider Roots",
          "tracks" : 941,
          "offline" : false,
          "index" : 7
        },
        {
          "type" : "playlist",
          "name" : "Coastsider Mark Knopfler",
          "tracks" : 261,
          "offline" : false,
          "index" : 8
        },
        {
          "type" : "playlist",
          "name" : "Wine Tasting",
          "tracks" : 28,
          "offline" : false,
          "index" : 9
        }
      ]
    },
    {
[... 781 lines cut ...]
      "index" : 130
    },
    {
      "type" : "playlist",
      "name" : "The Beatles",
      "tracks" : 67,
      "offline" : false,
      "index" : 131
    },
    {
      "type" : "playlist",
      "name" : "Tycho",
      "tracks" : 32,
      "offline" : false,
      "index" : 132
    },
    {
      "type" : "playlist",
      "name" : "Vocal Trance/Techno",
      "tracks" : 80,
      "offline" : false,
      "index" : 133
    }
  ]
}

skikirkwood, 2016-07-05:
I fixed this myself over the weekend.  Just got pulled into master with b2a214a commit.

apinner, 2016-07-06:
Can it be closed then?

closed 2016-07-06 by skikirkwood