volumio/Volumio2 #509: Can't Play User Created Playlists

skikirkwood, 2016-07-02:
I created a one song playlist and the file looks fine:

volumio@volumio2:/data/playlist$ ls
Classic British Rock
volumio@volumio2:/data/playlist$ more Classic\ British\ Rock 
[
  {
    "service": "mpd",
    "uri": "/NAS/iTunes/801/Live/06 Baby's On Fire.m4a",
    "title": "Baby's On Fire",
    "artist": "801",
    "album": "Live",
    "albumart": "/albumart?web=801/Live/large&path=NAS%2FiTunes%2F801%2FLive%2F06%20Baby's%20On%20Fire.m4a"
  }
]

But it will not play, with the log file showing:

2016-07-02T00:20:49.195Z - info: CoreCommandRouter::executeOnPlugin: mpd , handleBrowseUri
2016-07-02T00:20:49.196Z - info: [1467418849196] Listing playlists
2016-07-02T00:20:51.405Z - info: CoreCommandRouter::executeOnPlugin: mpd , handleBrowseUri
2016-07-02T00:20:55.190Z - info: CoreCommandRouter::volumioAddQueueItems
2016-07-02T00:20:55.190Z - info: {"uri":"/NAS/iTunes/801/Live/06 Baby's On Fire.m4a","title":"Baby's On Fire","service":"mpd"}
2016-07-02T00:20:55.191Z - info: [1467418855190] CoreStateMachine::addQueueItems
2016-07-02T00:20:55.191Z - info: [1467418855191] CorePlayQueue::addQueueItems
2016-07-02T00:20:55.192Z - info:  uri=/NAS/iTunes/801/Live/06 Baby's On Fire.m4a, title=Baby's On Fire, service=mpd
2016-07-02T00:20:55.192Z - info: First index is 0
2016-07-02T00:20:55.193Z - info: ADDING THIS ITEM TO QUEUE: {"uri":"/NAS/iTunes/801/Live/06 Baby's On Fire.m4a","title":"Baby's On Fire","service":"mpd"}
2016-07-02T00:20:55.193Z - info: Exploding uri /NAS/iTunes/801/Live/06 Baby's On Fire.m4a in service mpd
2016-07-02T00:20:55.194Z - info: ----------------------------/mnt//NAS/iTunes/801/Live/06 Baby's On Fire.m4a
2016-07-02T00:20:55.335Z - info: ----->>>>> {}
2016-07-02T00:20:55.336Z - info: [1467418855335] CorePlayQueue::saveQueue
2016-07-02T00:20:55.336Z - info: Adding item to queue: []
2016-07-02T00:20:55.337Z - info: CoreCommandRouter::volumioPushQueue
2016-07-02T00:20:55.338Z - info: [1467418855337] InterfaceWebUI::pushQueue
2016-07-02T00:20:55.338Z - info: []
2016-07-02T00:20:55.340Z - info: [1467418855340] CoreStateMachine::updateTrackBlock
2016-07-02T00:20:55.341Z - info: [1467418855340] CorePlayQueue::getTrackBlock
2016-07-02T00:20:55.343Z - info: An error occurred while exploding URI: TypeError: Cannot read property 'service' of undefined

volumio, 2016-07-07:
Yep, the tracks do have bad URI

skikirkwood, 2016-07-07:
Oops, meant to close a Spotify playlist issue I reported instead of this issue.

xipmix, 2016-07-10:
So I did some digging and it looks like things go wrong quite early, when reading the data back from the mpd plugin. I did the following:

patched app/plugins/music_service/mpd/index.js like so:

    diff --git a/app/plugins/music_service/mpd/index.js b/app/plugins/music_service/mpd/index.js
    index c350404..c43ef4d 100644
    --- a/app/plugins/music_service/mpd/index.js
    +++ b/app/plugins/music_service/mpd/index.js
    @@ -1003,7 +1003,7 @@ ControllerMpd.prototype.lsInfo = function (uri) {
                                for (var i = 0; i < lines.length; i++) {
                                        var line = lines[i];
     
    -                    //self.logger.info("LINE "+line);
    +                    self.logger.info("app/plugins/music_service/mpd/index.js sendCommand LINE "+line);
                                        if (line.indexOf('directory:') === 0) {
                                                path = line.slice(11);
                                                name = path.split('/').pop();
    @@ -1493,6 +1493,7 @@ ControllerMpd.prototype.handleBrowseUri = function (curUri) {
         var self = this;
     
         var response;
    +    self.commandRouter.logger.info("app/plugins/music_service/mpd/index.js ControllerMpd.prototype.handleBrowseUri curUri='"+curUri+"'");
     
         if (curUri.startsWith('music-library')) {
             response = self.lsInfo(curUri);
    @@ -1524,7 +1525,8 @@ ControllerMpd.prototype.explodeUri = function(uri) {
     
         if(uri.startsWith('search://'))
         {
    -        //exploding search
    +        self.commandRouter.logger.info("ControllerMpd.prototype.explodeUri search case, uri='"+uri+"'");
    +         //exploding search
             var splitted=uri.split('/');
     
             var argument=splitted[2];
    @@ -1601,6 +1603,9 @@ ControllerMpd.prototype.explodeUri = function(uri) {
                                     var name = path.split('/');
                                     var count = name.length;
     
    +        self.commandRouter.logger.info("ControllerMpd.prototype.explodeUri search album, line='"+line+"'");
    +        self.commandRouter.logger.info("ControllerMpd.prototype.explodeUri search album, path='"+path+"'");
    +
                                     var artist = self.searchFor(lines, i + 1, 'Artist:');
                                     var album = self.searchFor(lines, i + 1, 'Album:');
                                     var title = self.searchFor(lines, i + 1, 'Title:');

* deleted my only playlist
* browse to music library -> NAS -> an album -> a track
* from the options menu for the track select 'add to playlist'
* add track to new playlist 'test'.

What I see in the log is

    info: CoreCommandRouter::executeOnPlugin: mpd , handleBrowseUri
    info: app/plugins/music_service/mpd/index.js ControllerMpd.prototype.handleBrowseUri        curUri='music-library'
    info: app/plugins/music_service/mpd/index.js sendCommand LINE directory: NAS
    info: app/plugins/music_service/mpd/index.js sendCommand LINE Last-Modified: 2015-05-24T10: 51:44Z
    info: app/plugins/music_service/mpd/index.js sendCommand LINE
    info: CoreCommandRouter::executeOnPlugin: mpd , handleBrowseUri
    info: app/plugins/music_service/mpd/index.js ControllerMpd.prototype.handleBrowseUri       curUri='music-library/NAS'
    info: app/plugins/music_service/mpd/index.js sendCommand LINE directory: NAS/ABC_Kids
    info: app/plugins/music_service/mpd/index.js sendCommand LINE Last-Modified: 2013-04-21T06: 25:39Z
    ...
    info: app/plugins/music_service/mpd/index.js ControllerMpd.prototype.handleBrowseUri        curUri='music-library/NAS/ABC_Kids'

And here is where we have gone wrong. After this point the uri propagates in the form music-library/NAS/blabla instead of music-library/mnt/NAS/blabla.

I can't understand the code well enough to figure out where the /mnt should go in, but in another case it is added.

    app/plugins/music_service/mpd/index.js
    ....
    ControllerMpd.prototype.lsInfo = function (uri) { 
    ...
    self.clientMpd.sendCommand(cmd(command, []), function (err, msg) {
    ...
    else if (line.indexOf('playlist:') === 0) {
        path = line.slice(10);
        name = path.split('/').pop();
        if (path.endsWith('.cue')) {
            try {
                 var cuesheet = parser.parse('/mnt/' + path);

volumio, 2016-07-10:
Fixed

closed 2016-07-10 by volumio

ScotchTape, 2016-07-23:
For me it's not fixed. Did a git pull today:
root@volumio:/volumio# git log -1
commit c3908e504f7c3d333d1a671db206d031283406fb

but when I try to create a new playlist from an album, mounted via NFS and try to play that list:
Jul 23 13:31:02 volumio volumio[24193]: info: CoreCommandRouter::executeOnPlugin: mpd , getAlbumArt
Jul 23 13:31:02 volumio volumio[24193]: info: Pushing Favourites {"uri":"NAS/Musik/Kraftwerk/Minimum Maximum","favourite":false}
Jul 23 13:31:07 volumio volumio[24193]: --------------------------playlists
Jul 23 13:31:07 volumio volumio[24193]: info: CoreCommandRouter::executeOnPlugin: mpd , handleBrowseUri
Jul 23 13:31:07 volumio volumio[24193]: info: [1469280667861] Listing playlists
Jul 23 13:31:11 volumio volumio[24193]: info: [1469280671494] Play playlist minimax
Jul 23 13:31:11 volumio volumio[24193]: info: CoreCommandRouter::volumioClearQueue
Jul 23 13:31:11 volumio volumio[24193]: info: [1469280671506] CoreStateMachine::ClearQueue
Jul 23 13:31:11 volumio volumio[24193]: info: [1469280671509] CoreStateMachine::stop
Jul 23 13:31:11 volumio volumio[24193]: info: [1469280671512] CoreStateMachine::setConsumeUpdateService undefined
Jul 23 13:31:11 volumio volumio[24193]: info: [1469280671514] CorePlayQueue::clearPlayQueue
Jul 23 13:31:11 volumio volumio[24193]: info: [1469280671517] CorePlayQueue::saveQueue
Jul 23 13:31:11 volumio volumio[24193]: info: CoreCommandRouter::volumioPushQueue
Jul 23 13:31:11 volumio volumio[24193]: info: [1469280671523] InterfaceWebUI::pushQueue
Jul 23 13:31:11 volumio volumio[24193]: info: []
Jul 23 13:31:11 volumio volumio[24193]: info: CoreCommandRouter::volumioAddQueueItems
Jul 23 13:31:11 volumio volumio[24193]: info: [{"uri":"/NAS/Musik/Kraftwerk/Minimum Maximum/1-01 The Man Machine (Warsaw).m4a","service":"mpd"},{"uri":"/NAS/Musik/Kraftwerk/Minimum Maximum/1-02 Planet Of Visions (Ljubljana).m4a","s
Jul 23 13:31:11 volumio volumio[24193]: mum Maximum/2-10 Music Non Stop (Moscow).m4a","service":"mpd"}]
Jul 23 13:31:11 volumio volumio[24193]: info: [1469280671534] CoreStateMachine::addQueueItems
Jul 23 13:31:11 volumio volumio[24193]: info: [1469280671539] CorePlayQueue::addQueueItems
Jul 23 13:31:11 volumio volumio[24193]: info:  uri=/NAS/Musik/Kraftwerk/Minimum Maximum/1-01 The Man Machine (Warsaw).m4a, service=mpd, uri=/NAS/Musik/Kraftwerk/Minimum Maximum/1-02 Planet Of Visions (Ljubljana).m4a, service=mpd, u
Jul 23 13:31:11 volumio volumio[24193]: info: First index is 0
Jul 23 13:31:11 volumio volumio[24193]: info: ADDING THIS ITEM TO QUEUE: {"uri":"/NAS/Musik/Kraftwerk/Minimum Maximum/1-01 The Man Machine (Warsaw).m4a","service":"mpd"}
Jul 23 13:31:11 volumio volumio[24193]: info: Exploding uri /NAS/Musik/Kraftwerk/Minimum Maximum/1-01 The Man Machine (Warsaw).m4a in service mpd
Jul 23 13:31:11 volumio volumio[24193]: info: ----------------------------/mnt//NAS/Musik/Kraftwerk/Minimum Maximum/1-01 The Man Machine (Warsaw).m4a
Jul 23 13:31:11 volumio volumio[24193]: info: ADDING THIS ITEM TO QUEUE: {"uri":"/NAS/Musik/Kraftwerk/Minimum Maximum/1-02 Planet Of Visions (Ljubljana).m4a","service":"mpd"}
Jul 23 13:31:11 volumio volumio[24193]: info: Exploding uri /NAS/Musik/Kraftwerk/Minimum Maximum/1-02 Planet Of Visions (Ljubljana).m4a in service mpd
Jul 23 13:31:11 volumio volumio[24193]: info: ----------------------------/mnt//NAS/Musik/Kraftwerk/Minimum Maximum/1-02 Planet Of Visions (Ljubljana).m4a
Jul 23 13:31:11 volumio volumio[24193]: info: ADDING THIS ITEM TO QUEUE: {"uri":"/NAS/Musik/Kraftwerk/Minimum Maximum/1-03 Tour De France Etape One (Riga).m4a","service":"mpd"}
Jul 23 13:31:11 volumio volumio[24193]: info: Exploding uri /NAS/Musik/Kraftwerk/Minimum Maximum/1-03 Tour De France Etape One (Riga).m4a in service mpd
Jul 23 13:31:11 volumio volumio[24193]: info: ----------------------------/mnt//NAS/Musik/Kraftwerk/Minimum Maximum/1-03 Tour De France Etape One (Riga).m4a
Jul 23 13:31:11 volumio volumio[24193]: info: ADDING THIS ITEM TO QUEUE: {"uri":"/NAS/Musik/Kraftwerk/Minimum Maximum/1-04 Chrono (Riga).m4a","service":"mpd"}
Jul 23 13:31:11 volumio volumio[24193]: info: Exploding uri /NAS/Musik/Kraftwerk/Minimum Maximum/1-04 Chrono (Riga).m4a in service mpd
Jul 23 13:31:11 volumio volumio[24193]: info: ----------------------------/mnt//NAS/Musik/Kraftwerk/Minimum Maximum/1-04 Chrono (Riga).m4a
Jul 23 13:31:11 volumio volumio[24193]: info: ADDING THIS ITEM TO QUEUE: {"uri":"/NAS/Musik/Kraftwerk/Minimum Maximum/1-05 Tour De France Etape Two (Riga).m4a","service":"mpd"}
Jul 23 13:31:11 volumio volumio[24193]: info: Exploding uri /NAS/Musik/Kraftwerk/Minimum Maximum/1-05 Tour De France Etape Two (Riga).m4a in service mpd
[...]
Jul 23 13:31:11 volumio mpd[482]: db: No such directory
Jul 23 13:31:11 volumio volumio[24193]: info: ----->>>>> {}
Jul 23 13:31:11 volumio mpd[482]: db: No such directory
Jul 23 13:31:11 volumio mpd[482]: db: No such directory
Jul 23 13:31:11 volumio mpd[482]: db: No such directory
Jul 23 13:31:11 volumio mpd[482]: db: No such directory
Jul 23 13:31:11 volumio mpd[482]: db: No such directory
Jul 23 13:31:11 volumio mpd[482]: db: No such directory
Jul 23 13:31:11 volumio mpd[482]: db: No such directory
[...]
Jul 23 13:31:11 volumio volumio[24193]: info: ----->>>>> {}
Jul 23 13:31:11 volumio volumio[24193]: info: ----->>>>> {}
Jul 23 13:31:11 volumio volumio[24193]: info: [1469280671868] CorePlayQueue::saveQueue
Jul 23 13:31:11 volumio volumio[24193]: info: Adding item to queue: []
[... cut ...]
Jul 23 13:59:14 volumio volumio[24193]: info: [1469282354633] CorePlayQueue::getTrack 0
Jul 23 13:59:14 volumio volumio[24193]: info: CoreCommandRouter::volumioPushState
Jul 23 13:59:14 volumio volumio[24193]: info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
Jul 23 13:59:14 volumio volumio[24193]: info: [1469282354639] InterfaceWebUI::pushState
Jul 23 13:59:14 volumio volumio[24193]: info: Pushing Favourites {"service":"mpd","uri":"mnt/NAS/Musik/Kraftwerk/Minimum Maximum/1-11 Radioactivity (Warsaw).m4a","favourite":false

xipmix, 2016-07-24:
The fix for this went into 'master' at 34267cc4475bc69cf565e4b04e6a26f43e9fcceb on Sun Jul 10 20:37:47 2016 +0000 but for some reason this has not propagated to 'dev', which is still at 'c3908e5 Fixed credentials for NFS'. See #543 for some notes on how I updated my install.