volumio/Volumio2 #511: Scan of music library on network drive not completing

skikirkwood, 2016-07-02:
On RC2 Hotfix, I have two network drives mounted on my iMac as SMB shares.  After adding them the scanning process goes for some period of time and then stops.  The first scan stopped after 3000 tracks.  Trying a fourth restart after the last scan stopped at 16,000 tracks.  My library has about 20,000 tracks.

The scan just crashed my Raspberry Pi 3.  After rebooting the two network drives did not mount.  Log file below:

volumio@volumio2:/var/log$ cat volumio.log 
1970-01-01T00:00:21.969Z - info: -------------------------------------------
1970-01-01T00:00:21.976Z - info: -----            Volumio2              ----
1970-01-01T00:00:21.976Z - info: -------------------------------------------
1970-01-01T00:00:21.977Z - info: -----          System startup          ----
1970-01-01T00:00:21.977Z - info: -------------------------------------------
1970-01-01T00:00:22.236Z - info: Plugin folders cleanup
1970-01-01T00:00:22.237Z - info: Scanning into folder /volumio/app/plugins/
1970-01-01T00:00:22.238Z - info: Scanning category audio_interface
1970-01-01T00:00:22.239Z - info: Scanning category miscellanea
1970-01-01T00:00:22.240Z - info: Scanning category music_service
1970-01-01T00:00:22.241Z - info: Scanning category plugins.json
1970-01-01T00:00:22.242Z - info: Scanning category system_controller
1970-01-01T00:00:22.243Z - info: Scanning category user_interface
1970-01-01T00:00:22.244Z - info: Scanning into folder /data/plugins/
1970-01-01T00:00:22.245Z - info: Scanning category music_service
1970-01-01T00:00:22.246Z - info: Plugin folders cleanup completed
1970-01-01T00:00:22.248Z - info: Loading plugins from folder /volumio/app/plugins/
1970-01-01T00:00:22.256Z - info: Loading plugins from folder /data/plugins/
1970-01-01T00:00:22.261Z - info: Loading plugin "system"...
1970-01-01T00:00:22.319Z - info: Loading plugin "appearance"...
1970-01-01T00:00:22.681Z - info: Loading plugin "i2s_dacs"...
1970-01-01T00:00:22.692Z - info: Loading plugin "network"...
1970-01-01T00:00:22.767Z - info: Loading plugin "services"...
1970-01-01T00:00:22.772Z - info: Loading plugin "alsa_controller"...
1970-01-01T00:00:22.869Z - info: Loading plugin "volumio_command_line_client"...
1970-01-01T00:00:22.877Z - info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
1970-01-01T00:00:22.878Z - info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
1970-01-01T00:00:22.879Z - info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
1970-01-01T00:00:22.910Z - info: Loading plugin "volumiodiscovery"...
1970-01-01T00:00:22.980Z - info: Applying required configuration parameters for plugin volumiodiscovery
1970-01-01T00:00:23.026Z - info: CoreCommandRouter::executeOnPlugin: system , registerCallback
1970-01-01T00:00:23.027Z - info: Loading plugin "upnp"...
1970-01-01T00:00:23.035Z - info: [23034] Starting Upmpd Daemon
1970-01-01T00:00:23.051Z - info: CoreCommandRouter::executeOnPlugin: system , registerCallback
1970-01-01T00:00:23.052Z - info: Loading plugin "airplay_emulation"...
1970-01-01T00:00:23.058Z - info: [23058] Starting Shairport Sync
1970-01-01T00:00:23.061Z - info: Loading plugin "mpd"...
1970-01-01T00:00:23.119Z - info: Loading plugin "networkfs"...
1970-01-01T00:00:23.166Z - info: Loading plugin "alarm-clock"...
1970-01-01T00:00:23.381Z - info: Loading plugin "webradio"...
1970-01-01T00:00:24.144Z - info: Loading plugin "albumart"...
1970-01-01T00:00:24.164Z - info: Plugin example_plugin is not enabled
1970-01-01T00:00:24.165Z - info: Loading plugin "updater_comm"...
1970-01-01T00:00:24.365Z - info: Plugin mpdemulation is not enabled
1970-01-01T00:00:24.366Z - info: Loading plugin "websocket"...
1970-01-01T00:00:24.398Z - info: Loading plugin "spop"...
1970-01-01T00:00:25.186Z - info: Applying required configuration parameters for plugin spop
1970-01-01T00:00:25.196Z - info: ___________ START PLUGINS ___________
1970-01-01T00:00:25.202Z - info: CoreCommandRouter::volumioAddToBrowseSources[object Object]
1970-01-01T00:00:25.204Z - info: [25203] CoreMusicLibrary::Adding element Webradio
1970-01-01T00:00:25.229Z - info: Loading i18n strings for locale en
1970-01-01T00:00:25.239Z - info: MPD I18N LOAD FOR LOCALE en
1970-01-01T00:00:25.292Z - info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
1970-01-01T00:00:25.293Z - info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
1970-01-01T00:00:25.294Z - info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
1970-01-01T00:00:25.295Z - info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
1970-01-01T00:00:25.295Z - info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
1970-01-01T00:00:25.309Z - info: BOOT COMPLETED
1970-01-01T00:00:25.351Z - info: [25350] CoreStateMachine::resetVolumioState
1970-01-01T00:00:25.352Z - info: [25352] CoreStateMachine::getcurrentVolume
1970-01-01T00:00:25.353Z - info: CoreCommandRouter::volumioRetrievevolume
1970-01-01T00:00:25.403Z - info: [25401] CoreStateMachine::updateTrackBlock
1970-01-01T00:00:25.406Z - info: [25404] CorePlayQueue::getTrackBlock
1970-01-01T00:00:25.504Z - info: SpopD Daemon Started
1970-01-01T00:00:25.522Z - info: MPD running with PID493 ,establishing connection
[... 77 lines cut ...]
2016-07-02T00:43:11.076Z - info: mDNS: Found device Volumio2
2016-07-02T00:43:14.127Z - info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
2016-07-02T00:43:18.134Z - info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
2016-07-02T00:43:22.143Z - info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
2016-07-02T00:43:26.137Z - info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
2016-07-02T00:43:30.139Z - info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
2016-07-02T00:43:34.142Z - info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
2016-07-02T00:43:38.143Z - info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
2016-07-02T00:43:42.148Z - info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
2016-07-02T00:43:46.148Z - info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
2016-07-02T00:43:50.475Z - info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
2016-07-02T00:43:55.477Z - info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
2016-07-02T00:44:10.474Z - info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
2016-07-02T00:44:23.474Z - info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
2016-07-02T00:44:28.032Z - info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
2016-07-02T00:44:36.076Z - info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
2016-07-02T00:44:40.077Z - info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
2016-07-02T00:44:44.077Z - info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
2016-07-02T00:44:48.080Z - info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
2016-07-02T00:44:52.090Z - info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
2016-07-02T00:44:56.085Z - info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
2016-07-02T00:45:00.091Z - info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
2016-07-02T00:45:04.096Z - info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
2016-07-02T00:45:08.098Z - info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
volumio@volumio2:/var/log$

skikirkwood, 2016-07-03:
Update, I've hit Rescan several times, and got up to a little over 19,000 tracks, so still missing 2000 or so.  Hitting Rescan now doesn't seem to initiate a rescan although I do see the command actions in the log file.

apinner, 2016-07-04:
Do you know the file type that isn't scanning?

Is it mp4 or m4a with embedded artwork by any chance?

skikirkwood, 2016-07-04:
I don't believe there is any embedded artwork from what I can see, however when I delete the shares and start over, the scan seems to stop reliably at a Clannad folder, which has track names with special characters in them (Irish/Celtic), so perhaps that is causing an issue.

I'm setting up a dev environment, so I'll try to add some additional logging into the scanning process and try to narrow down the problem further.

apinner, 2016-07-04:
We had something similar here; https://github.com/volumio/Volumio2/issues/285

skikirkwood, 2016-07-04:
To test I will delete the Clannad folder and start a rescan from scratch, along with putting some additional logging in my dev environment to see exactly what track is killing it.

skikirkwood, 2016-07-08:
Here's an update.  I successfully got my entire music library scanned, but it took a total of 6 scans.  The last few times I just initiated the scan using "mpc update" in a terminal window.

After removing folders of music at and nearby where the first scan stopped I unmounted the network drive, remounted it and started a new scan from scratch.  The scan stopped at the same number of tracks as the previous time, although there was different music now where the scan stopped.  No embedded images, ordinary ALAC files.

After starting a new scan, it then stopped at the same number of tracks as the first round, but again, because I had moved out folders of music the stopping point of the last track scanned was different.

After 6 scans I got through all 21,011 tracks.

Another oddity is "mpc stats" shows a 21,011 tracks whereas Volumio's My Music shows 20,976.  Both show the same number artists and albums.

Volumio uses a custom build of MPD, right?  If so, adding a detailed logging function, if one isn't already there, would be very helpful.

apinner, 2016-07-08:
mpd was actually updated today to the latest version https://github.com/volumio/Build/commit/6b9a38607bb615ac632b83b546b35aa7c9580878

skikirkwood, 2016-07-08:
Thanks, grabbing it right now and will do a rescan from a new mount and see how it works.

skikirkwood, 2016-07-08:
Just grabbed the newest mpd but it's not starting.  Is there an easy way to launch it, or would it be better for me to build an entirely new image for my RPi?

volumio, 2016-07-08:
No it should start without any problem... Make sure you got the right version (arm or x86)

skikirkwood, 2016-07-08:
Ok, got it going.  Will remove my mounted network drive, mount it back, rescan and let's see if we can get to 21,000 tracks the first time.

volumio, 2016-07-08:
Fingersi crossed...

skikirkwood, 2016-07-08:
Unfortunately the new MPD didn't solve the problem of the scanning stopping prematurely.  Is stopped at exactly the same place - 3313 tracks.  I just did a new rescan and let's see if it stops at 16,137 tracks like the last time.

I started up MPD with verbose logging to see what is going on.  It generates a huge number of errors about metadata for just about all of my tracks, which I assume is fine.  Here's the tail end of the log when it stopped - no errors - seemed to think it was done.  BTW, no problems with Volumio 1.55 scanning the same network drive, and if I recall correctly, Volumio 2 RC1 had a successful one pass scan.

update: added NAS/iTunes/Clarence Clemons/Hero/07 Cross the Line.m4a
update: reading NAS/iTunes/Clarence Clemons/Hero/08 Kissin' On U.m4a
ffmpeg/mov,mp4,m4a,3gp,3g2,mj2: ISO: File Type Major Brand: M4A
ffmpeg/mov,mp4,m4a,3gp,3g2,mj2: Unhandled or malformed custom metadata of size 95
ffmpeg/mov,mp4,m4a,3gp,3g2,mj2: Unhandled or malformed custom metadata of size 154
ffmpeg/mov,mp4,m4a,3gp,3g2,mj2: Unhandled or malformed custom metadata of size 112
ffmpeg/mov,mp4,m4a,3gp,3g2,mj2: Unhandled or malformed custom metadata of size 149
ffmpeg/mov,mp4,m4a,3gp,3g2,mj2: All info found
update: added NAS/iTunes/Clarence Clemons/Hero/08 Kissin' On U.m4a
update: reading NAS/iTunes/Clarence Clemons/Hero/09 Christina.m4a
ffmpeg/mov,mp4,m4a,3gp,3g2,mj2: ISO: File Type Major Brand: M4A
ffmpeg/mov,mp4,m4a,3gp,3g2,mj2: Unhandled or malformed custom metadata of size 95
ffmpeg/mov,mp4,m4a,3gp,3g2,mj2: Unhandled or malformed custom metadata of size 154
ffmpeg/mov,mp4,m4a,3gp,3g2,mj2: Unhandled or malformed custom metadata of size 112
ffmpeg/mov,mp4,m4a,3gp,3g2,mj2: Unhandled or malformed custom metadata of size 149
ffmpeg/mov,mp4,m4a,3gp,3g2,mj2: All info found
update: added NAS/iTunes/Clarence Clemons/Hero/09 Christina.m4a
simple_db: removing empty directories from DB
simple_db: sorting DB
simple_db: writing DB
client: [2] process command "status"
client: [2] command returned 0
update: finished

apinner, 2016-07-08:
Bugger!

skikirkwood, 2016-07-09:
Good news, I found that when I removed the track alphabetically ahead of where the scan stopped the first time I got all the way to 16,737 tracks on the first scan.  It's an ALAC Tiesto track, not sure why it caused problems.

Having unmounted my network share and remounting it several times now to debug this, I now have a problem where after removing the network share and checking /mnt/NAS is empty, the iTunes folder mysteriously shows up and stops me from remounting the drive.  I think this was reported earlier.  A few reboots are in order, but it looks like the problem was due to issues with a few tracks, although MPD in Volumio 1.55 didn't have any issues with them.

apinner, 2016-07-09:
Could it be the ë in Tiësto thats causing the issue? Which would actually point towards the issue that i mentioned before.

skikirkwood, 2016-07-09:
That was my guess. After I figure out how to get not network drive mounted again I'll try some more scans. Didn't see any special characters around the tracks where it stopped at 16000 though.   
  

  
I could also create a new network library with these problematic tracks in it and use it as a test case.

skikirkwood, 2016-07-29:
Latest version works perfectly. Closing this.

closed 2016-07-29 by skikirkwood