volumio/Volumio2 #516: [RC2 hotfix] Spotify plugin only let's me browse playlists

pmd2016, 2016-07-03:
Hi!

Just did a clean install of Volumio RC2 (hotfix) and installed the Spotify plugin. The Spotify plugin only let's me browse and play my playlists. I can't browse or search for any other album, artist or song. Am I missing something? I do have a Premium (Family) account.

[image]

skikirkwood, 2016-07-06:
Browsing and playing playlists is the only functionality working in the new Spotify plugin.  We should have  a much richer feature set soon.

skikirkwood, 2016-07-15:
Ok, added "Features Playlists", "What's New" and "Genre's & Moods" to the Spotify plugin and will checkin the code this weekend to be pulled into a future RC.

We're currently using an SPOP module based on the old libspotify to get access to your playlists with authentication via userid and password.  There is no capability to get additional personal info such as your saved albums, tracks, etc.  The good news is the new Spotify Web API has incredibly rich metadata, which we used for the new features.  But to get personal user data you need to authenticate with an OAUTH workflow, so we'll have to wait a bit before we can add that.

The best news is the new Spotify API will allow for the creation of a Volumio2 user experience similar if not better in some ways to the official Spotify Web player, so stay tuned....

skikirkwood, 2016-07-29:
Download the latest version.  More to come.....

WalterBorgstein, 2016-08-28:
If beta testing is required I'd like to help!

skikirkwood, 2016-08-28:
Always could use help testing!  After a stable 2.0 release gets out I’ll add the Oauth stuff so we can access the personal data portion of the new Spotify API and get their saved tracks, albums, artists, etc. as well as let people modify their playlists with Spotify.  That will require a bit of testing!  I’ll let you know.

hoetmaaiers, 2016-12-01:
Curious how the Spotify integration into Volumio 2 is going. Is all the work done inside the Spotify plugin? I would like to contribute and benefit to a fully searchable and playable Spotify integration. What is already available, what is work in progress and what has yet to  come?

When in Volumio 2 with the spotify plugin I can only use Spotify through Spotify Connect. This makes me wonder what I am doing wrong?

skikirkwood, 2016-12-02:
@hoetmaaiers, there are two separate plugins for Spotify right now with Volumio2 - "Spotify" and a Spotify Connect module under development called volspotconnect at https://github.com/balbuze/volumio-plugins/tree/master/plugins/music_service/volspotconnect.

You should have the new features I listed above with the Spotify plugin - "Featured Playlists", "What's New" and "Genre's & Moods", plus integration with search.  We're working on developing a MyVolumio web site that will a bunch of new things, including OAuth Spotify integration.  With this working we'll be able to do the next phase of Spotify work, such as accessing your saved albums, tracks, artists, etc.

hoetmaaiers, 2016-12-04:
@skikirkwood is any help wanted on developing the new integrations?

skikirkwood, 2016-12-04:
Sure!  I just started a new job so haven’t had much time to start up the next phase yet.

The key thing to get working is OAuth workflow with the Spotify Web API.  It’s required before we can use the rest of the API calls that access a user’s personal info such as their saved tracks and albums.

I’m working with the Volumio team to create a new site, MyVolumio, that Volumio users can create accounts on.  We’ll be saving their Volumio playlists, favorites, settings, etc so they can easily restore these to any number of devices running Volumio2.  This is also where where we will have them do the OAuth workflow with their Spotify account, and get the OAuth token and refresh token to use in subsequent API calls.

For now, you could try to do an OAuth call without MyVolumio, get a Spotify token and refresh token, and just copy that somehow to a Volumio instance, maybe a config file or something.  With that, you could try making an API call that requires the token, such as accessing the user’s saved tracks.  Once we can do that we’ll be able to rapidly build out an experience similar to the Spotify Web music player, which is built on the new Spotify Web API.

Here is a good posting giving an overview of the Spotify Web API: https://labs.spotify.com/2015/03/09/understanding-spotify-web-api/ <https://labs.spotify.com/2015/03/09/understanding-spotify-web-api/>

And we’re using the Node.js wrapper for this library: https://github.com/thelinmichael/spotify-web-api-node <https://github.com/thelinmichael/spotify-web-api-node>

Note that in the current plugin we are using the old SPOP technology to retrieve a user’s playlists and also stream the audio.  With OAuth we can replace the former with a new Web API call, but will still rely on the SPOP daemon for music playback until Spotify releases something better to replace it.

Ping me with any questions, glad to have some help here!  I live in Palo Alto, California.

hoetmaaiers, 2016-12-12:
Hi @skikirkwood thanks for the update. Over the weekend I looked deeper into Spotify Connect and realised this is much more then a build in Airplay / Bluetooth thing build into the Spotify app. This makes it almost perfect for my purpose.

I still am interested in helping to bring Spotify Web API integration to Volumio, any todo board to collect and organise work?

skikirkwood, 2016-12-13:
Hi @hoetmaaiers, the Volumio team likes Trello, which we are using for MyVolumio. I personally prefer Asana, I think the UI is more refined and easier to use.  Do you have a preference?

hoetmaaiers, 2016-12-14:
I have no preference in those 2 tools, the way they are taken in action is the most important I think.
Can I get access to this board or is public access available?

Beste groet, 

Robin Houdmeyers
--
E robin.houdmeyers@gmail.com 
M +32 472 494 899

skikirkwood, 2016-12-14:
>  E  robin.houdmeyers@gmail.com (mailto:robin.houdmeyers@gmail.com)
>  M  +32 472 494 899 (tel:+32%20472%20494%20899)

Let me check with the 2 I am working with. Probably best to use a single system.       
  
  

  
  

  
BTW, we're doing MyVolomio on top of Drupal. Do you have any experience with it?
  

  
Drupal is PHP-based, and the world's largest open source software project. It should make it easy to accomplish our objectives here.

closed 2017-06-02 by volumio