volumio/Volumio2 #517: Volumio 2 RC2 0.928 Static IP Address not fixed

chsims1, 2016-07-03:
Setting a static IP address from UI apparently works initially, but before the boot is complete it is dropped and a dhcp request is sent for another address.  I've attached the 'journalctl -b' file where this can be clearly see (static 192.168.2.221 is changed to dynamic 192.168.2.19).  Loads of other errors shown 
[journalctl.txt](https://github.com/volumio/Volumio2/files/345105/journalctl.txt)
in that file connected with samba and other stuff.

mundodisco8, 2016-07-06:
Same (I think) here. Setting the fixed IP for my wireless adapter works: I had previously a DCHP-assigned address (X.X.X.220), and it changed to the fixed IP I wanted (X.X.X.11). The next day, the address was a new one (X.X.X.102). 

The funny thing is that I reinstalled the image and the same happened, the next day, the address would be the 102 one. I say that is funny, because it didn't switched back to the 220. 

etc/network/interfaces shows that the wlan adapter is in static mode, with the address that I wanted (the 11 one), but I can only access to the web server with the 102 one.

f there is something else that you think I can try to get more info about the issue, just tell me. I'm not an expert in linux, so I can't diagnose it, but I know my way around.

chsims1, 2016-07-07:
There is something funny going on here.  There is some script (ifup possibly) running which tinkers with wlan static vs dynamic addresses, so in your case you are setting the static one and the script comes along and also dynamically sets one.  Unfortunately I don't know how Volumio is handling networking.  Things changed last year with the way Raspbian works vs standard Debian.  It is not clear to me which methods Volumio is using.  All I know for certain is that the standard Debian was of setting a static IP through the interfaces file (which works fine in Volumio 1.55) does not work in Volumio 2.

volumio, 2016-07-07:
Yes, it behaves quite weirdly... And standard debian way does not work anymore... So I'm clueless at this point...

apinner, 2016-07-07:
http://askubuntu.com/questions/486474/system-loses-static-ip-address

I know its Ubuntu, but the symptoms seem to be the same. Can anyone try uncommenting the alias section of /etc/dhcp/dhclient.conf and adding a static address.

chsims1, 2016-07-07:
I have read in several places that the way Raspbian is supposed to handle static IP addresses now is via the dhcpd.conf not interfaces file. I don't think that package is installed with volumio??

apinner, 2016-07-07:
Yeah there was a comment along those lines on the Ubuntu thread;

> Interesting fix, when my NIC went back to DHCP, this created an alias NIC with the correct IP and my clients were none the wiser and everything hummed along. Still not sure why this is happening, but this fix will do for now!

Seems it does just that and creates an alias.

chsims1, 2016-07-08:
Yes it does work, but it doesn't solve my problem.  One of the reasons for wanting a static address is that my NAS needs the IP address of the client to allow a directory to be mounted as NFS.  Mounting doesn't work (for some reason) unless both of the IP addresses (static and dynamic) are permitted .... and of course I don't know the dynamic address in advance.

Can anyone tell me why dhclient is being used rather than the "official" Raspbian use of dhcpcd?  Use of this package would presumably solve the problem??

apinner, 2016-07-08:
Oh.. That's a shame, thought we had cracked it.

Don't dhclient and dhcpcd essentially do the same thing? The only difference being dhclient is a one shot deal, once it has obtained an IP it does nothing else. So i'm assuming it was chosen to remove overhead from the OS as its supposed to be as light as possible. Though, i could be barking up the wrong tree and it was just an oversight.

chsims1, 2016-07-08:
Yes, they do the same thing, and the one shot thing is probably correct.  However, if Raspbian is using dhcpcd, then presumably it is likely that there are further networking scripts tied in/customized towards it?

apinner, 2016-07-08:
From what i have read switching to isc-dhcp-client would be even better as it is more flexible. dhcpcd is all or nothing.

chsims1, 2016-07-08:
The way dhclient is being run in volumio, it isn't actually acting as a one-shot service:

> man dhclient

>-w Continue running even if no broadcast interfaces were found. Normally DHCP client will exit if it isn't able to identify any network interfaces to configure. (...)

I think I will remove dhclient and try using dhcpcd.  If nothing else it will confirm that dhclient doesn't work well (for static IP) in the current setup.

xipmix, 2016-07-09:
@chsims1 - can you try the following exercise- it will be easier to debug if you use the ethernet interface, not wifi.
 * set interface to auto IP address.
 * power cycle the machine
 * copy /etc/network/interfaces, to e.g. /home/volumio/setp1
 * set static address in UI. 
 * copy /e/n/i to /home/volumio/step2
 * grep -rl  192.168.1.13  /data   (use whatever IP address you set),
 * copy the files that the grep finds
 * reboot
 * copy /e/n/i to /home/volumio/step3
 * grep -rl  192.168.1.13  /data
 * copy files it turns up

Go through the files to see what has changed.

I tried this and found that volumio correctly updates the /etc/network/interfaces file (step2), so the necessary state should be saved on disk for it to work correctly on the next boot. The UI also showed the static address setup when I went back to it after the reboot.

xipmix, 2016-07-09:
It would also be useful to know what version you are on;

    cd /volumio; git log |grep commit |head -1

You can update with

    cd /volumio; git pull

chsims1, 2016-07-09:
Factory reset, auto IP for both eth0 and wlan0.

Step 1: initial settings (file step1 showing initial /etc/network/interfaces)
Step 2: set static IP for eth (file step2 showing /etc/network/interfaces)

```
volumio@volumio:~$ diff -y step1 step2

auto wlan0                                                      auto wlan0
auto lo                                                         auto lo
iface lo inet loopback                                          iface lo inet loopback

allow-hotplug eth0                                              allow-hotplug eth0
iface eth0 inet dhcp                                          | iface eth0 inet static
                                                              > address 192.168.2.221
                                                              > netmask 255.255.255.0
                                                              > gateway 192.168.2.1

allow-hotplug wlan0                                             allow-hotplug wlan0
iface wlan0 inet manual                                       | iface wlan0 inet static
                                                              > address undefined
                                                              > address undefined
```
So, static IP setting go into interfaces file, and this remains after rebooting (Step 3). Why does Volumio alter the wlan0 setting though???

ifconfig for steps 1 & 2 shows that the UI settings are working in practice (192.168.2.221 static IP, 192.168.2.19 dynamic for wlan0))

```
volumio@volumio:~$ diff -y eth1.ifconfig eth2.ifconfig

eth0      Link encap:Ethernet  HWaddr b8:27:eb:95:95:10         eth0      Link encap:Ethernet  HWaddr b8:27:eb:95:95:10  
          inet addr:192.168.2.17  Bcast:192.168.2.255  Mask:2 |           inet addr:192.168.2.221  Bcast:192.168.2.255  Mask:
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1              UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:6027 errors:0 dropped:0 overruns:0 frame |           RX packets:6762 errors:0 dropped:0 overruns:0 frame
          TX packets:535 errors:0 dropped:0 overruns:0 carrie |           TX packets:982 errors:0 dropped:0 overruns:0 carrie
          collisions:0 txqueuelen:1000                                    collisions:0 txqueuelen:1000 
          RX bytes:924118 (902.4 KiB)  TX bytes:167432 (163.5 |           RX bytes:1016769 (992.9 KiB)  TX bytes:412420 (402.

lo        Link encap:Local Loopback                             lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0                             inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1                        UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:7016 errors:0 dropped:0 overruns:0 frame |           RX packets:7653 errors:0 dropped:0 overruns:0 frame
          TX packets:7016 errors:0 dropped:0 overruns:0 carri |           TX packets:7653 errors:0 dropped:0 overruns:0 carri
          collisions:0 txqueuelen:1                                       collisions:0 txqueuelen:1 
          RX bytes:648963 (633.7 KiB)  TX bytes:648963 (633.7 |           RX bytes:711181 (694.5 KiB)  TX bytes:711181 (694.5

wlan0     Link encap:Ethernet  HWaddr 7c:dd:90:b0:de:c5         wlan0     Link encap:Ethernet  HWaddr 7c:dd:90:b0:de:c5  
          inet addr:192.168.2.19  Bcast:192.168.2.255  Mask:2             inet addr:192.168.2.19  Bcast:192.168.2.255  Mask:2
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1              UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:6213 errors:0 dropped:0 overruns:0 frame |           RX packets:6748 errors:0 dropped:0 overruns:0 frame
          TX packets:639 errors:0 dropped:0 overruns:0 carrie |           TX packets:698 errors:0 dropped:0 overruns:0 carrie
          collisions:0 txqueuelen:1000                                    collisions:0 txqueuelen:1000 
          RX bytes:1014815 (991.0 KiB)  TX bytes:115347 (112. |           RX bytes:1134457 (1.0 MiB)  TX bytes:128152 (125.
```
Routing tables identical for Step 1 and Step 2

```
volumio@volumio:~$ cat step2.netstat

Kernel IP routing table
Destination     Gateway         Genmask         Flags   MSS Window  irtt Iface
0.0.0.0         192.168.2.1     0.0.0.0         UG        0 0          0 eth0
192.168.2.0     0.0.0.0         255.255.255.0   U         0 0          0 eth0
```
and config data is being saved:

```
volumio@volumio:~$ cat step2.-data-configuration-system_controller-network-config.json

{
[... 67 lines cut ...]
allow-hotplug eth0
iface eth0 inet static
address 192.168.2.221
netmask 255.255.255.0
gateway 192.168.2.1

allow-hotplug wlan0
iface wlan0 inet static
address 192.168.2.222
netmask 255.255.255.0
gateway 192.168.2.1
```
So, the tests just show what I have described previously (apart from static IP on eth0 working fine). My router indicates that a static IP for wlan0 is initially assigned which is fairly quickly dropped for a dynamic one.

Version being used:

```
volumio@volumio:/volumio$ git log -1
    
commit 46580c4e728879e79625af1efac8ddd02b36e2d6
Author: volumio <info@volumio.org>
Date:   Thu Jun 30 21:43:15 2016 +0000

    fixed webradio my webradios
```

chsims1, 2016-07-11:
Not a solution to the problem directly, but I can confirm that removing the isc-dhcp-client and replacing it with the dhcpcd5 package allows my wlan0 to be set with a static IP address (I haven't tried if dhcp on eth0 works because I'm using a RPi Zero).

```
/etc/network/interfaces

auto wlan0
auto lo
iface lo inet loopback

allow-hotplug eth0
iface eth0 inet manual

allow-hotplug wlan0
iface wlan0 inet manual
```
```
/etc/dhcpcd.conf

interface wlan0                                                                                                                                                                    
static ip_address=192.168.2.222/24                                                                                                                                                 
static routers=192.168.2.1                                                                                                                                                         
static domain_name_servers=8.8.8.8 
```

closed 2017-06-02 by volumio