volumio/Volumio2 #520: Add System Information System Page

apinner, 2016-07-05:
Add information like CPU load, Temp, CPU Frequency, Memory Usage, SD card Size and Usage to the System page.

volumio, 2016-07-07:
We can do that, but it will make the system page slower (retrieving all this infos will take a while...

apinner, 2016-07-07:
Why can't it work like the Network page? We display where the information will be, and it shows once it is available. That way the current information will be there instantly but the extra info will take a little longer to load.

volumio, 2016-07-07:
Yep let's do that, maybe next week? Want to debug a few things...

apinner, 2016-07-07:
Yeah no rush.. I just thought it would be helpful for debugging. Especially as i was having issues the other day and some system information in the UI would have been useful.

AlexisFinn, 2017-11-22:
+1 for this feature. 

Just a thought, maybe it could live along a live log tool (that would display journalctl -f output) in a dev page (akin to /DEV) making it accessible for debugging while not immediately cumbersome in the interface, as I don't believe most people are concerned about cpu/ram/bandwith usage as long as everything seems to be working.

One thing that I believe everyone would be interested in knowing (so including non-technical users) and that shouldn't use up any extra resources is the available disk space. That could easily live pretty much anywhere as it doesn't need to be update more than say once every couple of minutes.