volumio/Volumio2 #524: RC2 [HOTFIX] deleted all my files

xarath, 2016-07-06:
Fresh install of RC2 [HOTFIX].  Mounted an NFS drive through the GUI and tried playing some music.  Then, I cleared the queue (and it wouldn't clear).  Files were no longer listed under NAS and I looked on the drive itself and every files were gone but the ones that were in the queue.  I had to restore from backups.  I would strongly advise anyone to mount read only.

jomo, 2016-07-06:
Likely relevant commit: 1d536ec6d75d894f7333cf1e71649e0abcaa9ed7

If I read that code correctly, it tries to `unmount` the NAS and if that fails, it just tries deleting it using `rm -rf`.

jomo, 2016-07-06:
nvm, it actually does that only if `umnount` results in `error === null`.

jomo, 2016-07-06:
I'm not sure if `rm -rf` was the culprit in this case (code looks good IMO), but #525 makes sure at least that can't happen.

volumio, 2016-07-06:
Oh bad thing... I prevented that by mounting read only, but I realize now that I did only for CIFS. Do you know how can we add the ro option for nfs?

xarath, 2016-07-06:
By adding ro option in /etc/fstab. Here is the relevant line:
192.168.1.5:/data3/music	/mnt/NAS/Music2	nfs4	_netdev,auto,ro	0	0

volumio, 2016-07-06:
Should be solved with 
https://github.com/volumio/Volumio2/commit/d64213f9c050bb05542a2a4ae53656ecd2f056d1

Is anyone able to test by doing 

cd /volumio
git pull
killall node 

Please let me know

xipmix, 2016-07-07:
I don't think d64213f is the right fix, because it uses options that are not supported for NFS mounts.
I noticed this back in volumio1 but never figured out how to report it concisely. I think if you set the base options as "ro,noauto"  it will work better.
`
fsopts = credentials + "ro,noauto";
`
`
if (options) {
  fsopts = fsopts + "," + options;
}
`

You might want to try 'bg' as well so it will keep retrying but I'm not sure how well supported it is.

xipmix, 2016-07-07:
Also you may want to be explicit that you want 'soft' mounts `"ro,soft,noauto"`, which should be ok as long as the mount is readonly. If options contains 'rw' that might override and open the way to possible file corruption on the NAS end.

volumio, 2016-07-07:
Ok, added with 
https://github.com/volumio/Volumio2/commit/4a30ebaca97fb2e20be9a15b497bca17de2b0466

Could you please try if that works ok?

xipmix, 2016-07-07:
I tried it just now. volumio does not mount the NAS but a manual mount command works. Where should I look in the logs please?

xipmix, 2016-07-07:
Found it. journalctl -p 7 shows me the mount attempt:
 `mount -t nfs -o undefinedro,soft,noauto ...`

xipmix, 2016-07-07:
Fixed by setting `credentials = ''` in the NFS case.

volumio, 2016-07-07:
Fixed here https://github.com/volumio/Volumio2/commit/c3908e504f7c3d333d1a671db206d031283406fb

Could you try?

xipmix, 2016-07-07:
Shortly after I ran 'killall node' I got the startup sound and 'mount|grep nfs' showed the right thing.
But I was hasty and typed reboot to check that it was really working (I'd done a manual mount before). systemd made me wait for mpd to finish with the filesystem (around 1 minute), but eventually the reboot went ahead and voila the NAS filesystem is mounted 'ro'.

volumio, 2016-07-07:
Well done! Thank you very much! This was really urgent to solve...

closed 2016-07-07 by volumio

xipmix, 2016-07-07:
Just to cover this since it was mentioned by @xarath, the `_netdev` option will not hurt but is probably not needed now since systemd knows about NFS mounts. 

See for example http://codingberg.com/linux/systemd_when_to_use_netdev_mount_option

jomo, 2016-07-07:
@volumio it's still not clear why the files were deleted in the first place, is it?

volumio, 2016-07-07:
Yes, we know why. NFS was not mounted Read only, and when the share was deleted, mpd was updating the db. So the share was not unmounteed, but the folder was deleted. 
With this fix it won't happen again, since the mount will be read only

jomo, 2016-07-07:
Shouldn't `unmount` return an error when it fails (so `rm -rf` would not run)?
Is this due to the lazy `unmount -l`? The man page says the filesystem is detached immediately though.

volumio, 2016-07-07:
Yes, it should do like that ... But if mpd is somehow scanning, this might take a while... Or at least this is my explanation...