volumio/Volumio2 #526: RC2 [HOTFIX] Playlist issues

SonosKiller, 2016-07-06:
Running a Rpi3 a clean install of the RC2 [Hotfix] on a new Sd card using no Dac just the audio jack
and a windows share as Nas

Playlist created from Music Library, Nas, ... don't Play from Playlists , Add to queue don't work either.
Playlist created/saved from  Queue both Play and Add to queue works fine from Playlists

Random play of the queue don't work when "Consume playlist" is active 

Thanks for the work and dedication you all put in to this software
May God bless you all

apinner, 2016-07-07:
Sorry I don't quite follow the first part? Playlists don't play in playlists? What can't you add to queue?

Can you clarify what you mean a bit better?

xipmix, 2016-07-07:
Sounds like #509 to me...

SonosKiller, 2016-07-07:
Sorry ! "Playlists dialoge"

If I create a playlist from Music Library, Nas, .. dialogue (and also in  search dialoge)
That playlist don't start when I pick them up in the Playlists dialogue using Play and  the playlist don't end up in queue if I use Add to queue

Playlists created/saved in Queue dialouge works fine when I pick them up in Playlists dialoge

Ok ?

Will test to  modify , add to and change them two types of playlists tonight and see what's happen

volumio, 2016-07-07:
IMHO they miss the /mnt/ in the URI

SonosKiller, 2016-07-07:
I assume that this also is a easy fix !
"Random play of the queue don't work when "Consume playlist" is active"

"Consume" is this the right word to use ? modify or change seems more correct ! :)

xipmix, 2016-07-07:
> 
> "Consume" is this the right word to use ? modify or change seems more correct ! :)

I noticed this too and wrestled with the wording.
What the button does is turn on the mode that does
'Remove items in the queue after they have been played'.
but that's a little long for a tool tip.

'Consume queue' is more consistent (its The Playqueue, not a playlist)
but...ugly.

My suggestion: 'Remove played items' or 'Autoremove'.

SonosKiller, 2016-07-07:
Okey ! Didn't notice that songs disappeared, thought that function was for activate "moving songs in the queue" , then Consume is more than correct

apinner, 2016-07-07:
Yeah Consume mode means it will consume the queue, so as a song is played it will be consumed and removed from the current track queue.

SonosKiller, 2016-07-07:
Just a note on the "Consume playlist" function (now when I understand it  :) ) , it removes the top song not the played song !
 Random and Consume together could be a nice feature

volumio, 2016-07-10:
Fixed

closed 2016-07-10 by volumio