volumio/Volumio2 #527: Default startup volume always start at 100%

Callenberg, 2016-07-07:
Even if I set the default start up volume to another value or even disable it in Settings: PLAYBACK OPTIONS/Volume Options the volume is always max, 100%, after reboot or restart.

 Best regards, Cbg

----
SW + HW:
Volumio 2 RC2 HOTFIX 0.928 Jun 30 23:50:57
Windows 10 Professional + Google Chrome Version 51.0.2704.103 m (64-bit)
Raspberry Pi 3
IQaudio Pi-DigiAMP+    19V 3.42A
Class 6 8GB SD card

Callenberg, 2016-07-09:
The bug is also reported by on the forum by Dennisf1:
[https://volumio.org/forum/volumio-rc2-t4288-120.html] 
(https://volumio.org/forum/volumio-rc2-t4288-120.html)

Can it be the case that Volumio regard the IQaudio Pi-DigiAmp as a DAC?

Callenberg, 2016-07-09:
. . . and gkkpch on the forum is promising an updated hotfix for the weekend. Great.
Thanks a lot.

volumio, 2016-07-10:
should be fixed with https://github.com/volumio/Volumio2/commit/0b4228b5cfa80829c70a4242855e3f74e508d12f

Callenberg, 2016-07-11:
Excellent!

apinner, 2016-07-12:
Fixed so closing

closed 2016-07-12 by apinner

fx789, 2016-09-07:
Is this fixed in the official download "volumio-0.979-2016-08-20-pi.img"?