volumio/Volumio2 #528: Activating Gapless breaks something and doesn't stay active

nightflightdan, 2016-07-07:
All set up and working fine. (pi3 with IQAudioDAC+)
Whilst playing music from NAS, going into settings, turn on gapless playback and click save, music stops, but the display still acts as if it's playing, hit f5, appears paused, clicking play brings up the "now playing" notification, but nothing comes out of the speakers.
checking in the settings and the gapless playback toggle is back at "off"
Requires restart to get it working again, gapless playback still at "off" after restart

Retry activating gapless playback... still no confirmation of settings being changed after clicking save, 
clicking play brings up the playback started notification but again, no music.
Navigate around a few other menus then back to playback options... gapless playback toggle is back at 'off'
retry activating gapless playback, hit save, hit play, now playing notification comes up but no sound and play icon stays as a triangle, click 'forwards' next track in playlist begins to play.

futuralogic, 2016-07-09:
I can repro this. Even starting from a fresh boot, no playback.  Go to Playback Options, set Gapless Playback ON - hit Save.  Nothing seems to happen.  Interestingly, if I change another setting in the same section, such as Audio Buffer Size, it appears to save it.

Raspberry Pi Model B.

To try to get the feature to actually work I tried manually modifying mpd.conf.  This saved, I rebooted, and it stuck.  However, there is no gapless playback.  I'm not sure why.

volumio, 2016-07-10:
I disabled gapless playback, since with the new queue we need to completely rewrite some code to make it work

closed 2016-07-10 by volumio