volumio/Volumio2 #531: RC2 2016-06-26 on ODroid XU4

from9to5, 2016-07-08:
usb 2.5 HDD with 25000 flac files, static IP over LAN, usb ODAC Ref B:
The database is updating, but stops after exactly 2500 titles.  I tried with another usb 3.5 HDD with external power but it was the same. I deleted one third of the HDD, but the database again stopped at exactly title 2500. If I disconnect my HDD und connect my NAS via Fritz!Box with the same music data, the database is working fine, every title is stored and can be played. By the way, I have the same behaviour with the Cubietruck regarding the file database via usb.

apinner, 2016-07-08:
Can you please give your issues a more descriptive title, having two issues with the same title isn't very helpful when we look through them.

skikirkwood, 2016-07-08:
I think this is the same issue as 511 that I reported.  The scan of my network drive seems to stop at the same points, even when I remove folders of music that were near where the scan stopped.

from9to5, 2016-07-08:
My NAS is OK. I have the same flac files on usb drive and NAS. But it won't read all the files if I connect the usb drive.

volumio, 2016-07-10:
Should be fixed with https://github.com/volumio/Volumio2/commit/b73ecb8be6f1e0ef12cbfba66a393bd9e621ee3d

apinner, 2016-07-12:
Fixed, so closing

closed 2016-07-12 by apinner