volumio/Volumio2 #532: RC2 2016-06-26 on ODroid XU4

from9to5, 2016-07-08:
NAS with 25000 flac files, static IP over LAN, usb ODAC Ref B:
The output audio level of the DAC was much too low and very noisy. I checked the alsamixer and the pcm level was only around 50% what gives only 200mV maximum Output at chinch. I adjusted the output level in alsamixer to 100% with curser up. It was not possible to store that. So after reboot I have to start putty and set the output level to 100% to get 2Vpp for my amplifier. That is annoying.
I had the same problem with my CubieTruck, but that one stored the level until I changed the usb DAC. So is it possible to create a config file, where we can set a parameter for output level? For most of the DACs is 100% the correct parameter. But I learnt from some people, that there are some DACs which will go into distortion and therefore they have to be driven at -3dB. So, the best imho is to create a parameter for maximum sound level in dB or %.

gkkpch, 2016-07-09:
i will have a look at this issue tonight after I'm finished with the new kernel for the C1

from9to5, 2016-07-16:
I cant´t give a feedback, cause I don´t know how to change the files on my ODroid XU4. I have to wait until someone crates anew image for ODroid 3/XU4.
Thanks anyway.

gkkpch, 2016-07-16:
Currently I am testing C1 and C2, XU4 and your issue is next on the list but can take a while. You might a newer image first without the fix. Don't worry, it will be taken care off.

closed 2017-06-02 by volumio