volumio/Volumio2 #534: Cannot add radio station

povserok, 2016-07-09:
Could you please test if you can add and then play radio stations? None of the radio stations I add work, but the URLs work on my PC. I have tried with the following:
http://bbcmedia.ic.llnwd.net/stream/bbcmedia_radio1_mf_p
http://mp3.rtvslo.si/rsi

Are there any logs  to see what is going on (apart from /var/log/volumio.log which does not give me any info why the radio is not working)?

apinner, 2016-07-09:
If you are adding BBC stations then they will not work at the moment due to how the BBC now encode their streams.

We updated MPD yesterday so BBC stations can now play. That will be in the next release.

povserok, 2016-07-09:
Thanks, can you please also check these streamss, if the issue is the same?
http://mp3.rtvslo.si/rsi
http://mp3.rtvslo.si/val202

Thanks.

volumio, 2016-07-10:
They now work, we'll release the update in few days...

closed 2016-07-10 by volumio