volumio/Volumio2 #540: AppleTV Audio via AirPlay

paviro, 2016-07-11:
I used to use Runeaudio as an AirPlay speaker for my AppleTV. I saw Volumio now supports synchronized AirPlay so I switched but sadly it seems like it now does not work at all. The Apple TV directly switches back to paused when you start playing something with Volumio selected as a speaker.

volumio, 2016-07-11:
What device and DAC do you have ?

paviro, 2016-07-11:
RPI2 and a HiFiBerry DAC+ Standard.

volumio, 2016-07-11:
Are you connected with cable or LAN?

paviro, 2016-07-11:
Both Apple TV and Pi are connected via WLAN.

volumio, 2016-07-11:
Could you try connecting pi to cable and see if that works?

paviro, 2016-07-11:
Does not help, any way I can provide you with some debug data?

volumio, 2016-07-11:
Yes, please paste the output of 
 sudo journalctl | grep shairport

jomo, 2016-07-11:
Does [this article](http://www.appletvhacks.net/2015/09/07/ios-9-and-os-x-el-capitan-wont-let-you-airplay-to-jailbroken-apple-tv-or-the-one-running-older-firmware/) help?

paviro, 2016-07-11:
Don`t think so since it worked before with Runeaudio...
```
volumio@volumio:~$ sudo journalctl | grep shairport
Jan 01 00:00:35 volumio shairport-sync[674]: startup
Jul 11 10:33:13 volumio shairport-sync[674]: Updating DB (#1) ...
Jul 11 10:33:13 volumio shairport-sync[674]: volume:100%   repeat: off   random: off   single: off   consume: off
Jul 11 10:33:17 volumio shairport-sync[674]: Updating DB (#1) ...
Jul 11 10:33:17 volumio shairport-sync[674]: volume:100%   repeat: off   random: off   single: off   consume: off
```

Also the status LED of my DAC turns on for a few seconds after pressing play.

volumio, 2016-07-11:
Ok, let's try this.

killall shairport-sync
nano /etc/shairport-sync.conf

change log_verbosity = 2;

close it. 
Start it with

/usr/local/bin/shairport-sync

and paste here the output...

paviro, 2016-07-11:
```
culling threads.
one joined...
new RTSP connection.
    CSeq: 36.
    X-Apple-Device-ID: 0x705681d6b74b.
    Apple-Challenge: nJdXBKgjggW28jW/DiVJxQ==.
    DACP-ID: E43FB08F535B18F4.
    Active-Remote: 2404776728.
    User-Agent: AirPlay/220.68.
    Content-Length: 0.
    Apple-Response: ANgS8zc7LQUmLAGVNAjgpJP3ByYKHlLP170FzbuLjM/5+LBqgMCAe84H81S1IM/s5TJ5tOHswTn2N/YO504ZJsZdKtk26JT8V6pF9xVJ5NecoBhURFrGy3ffUKbnoMY2MnU9jcsMjaBV+dcJSFWJgkhADMUHBqNpR20SbXqnNxiWWWPoY663bzTEseUWX4mAmxpmCa9FYLWr6RK4CPUVgn7ZcC6I5NfWpLSd9SGGRkzkj85I7ONmdWMFMUfa9UpFJt1FQwgxpQqQ+bC6hG5vLveqxRO8tjc+KvqcIsVQcLlpLmNZ3cPpp0OLU4wlhnrQ2r9eY85xhwdqzZGjLWKLpQ.
    CSeq: 36.
    Audio-Jack-Status: connected; type=analog.
    Public: ANNOUNCE, SETUP, RECORD, PAUSE, FLUSH, TEARDOWN, OPTIONS, GET_PARAMETER, SET_PARAMETER.
    X-Apple-Client-Name: Fernseher.
    CSeq: 37.
    X-Apple-Device-ID: 0x705681d6b74b.
    DACP-ID: E43FB08F535B18F4.
    Active-Remote: 2404776728.
    User-Agent: AirPlay/220.68.
    Content-Length: 646.
    Content-Type: application/sdp.
Play connection from device named "Fernseher".
Play connection from user agent "AirPlay/220.68".
Process metadata with type 73736e63, code 736e616d and length 9.
    CSeq: 37.
    Audio-Jack-Status: connected; type=analog.
Process metadata with type 73736e63, code 736e7561 and length 14.
    Transport: RTP/AVP/UDP;unicast;mode=record;timing_port=57755;x-events;control_port=59132.
    CSeq: 38.
    X-Apple-Device-ID: 0x705681d6b74b.
    DACP-ID: E43FB08F535B18F4.
    Active-Remote: 2404776728.
    User-Agent: AirPlay/220.68.
    Content-Length: 0.
Active-Remote string seen: "2404776728".
DACP-ID string seen: "E43FB08F535B18F4".
Process metadata with type 73736e63, code 61637265 and length 10.
User-Agent is AirPlay; selecting the AirPlay latency of 88200 frames.
Process metadata with type 73736e63, code 64616964 and length 16.
rtp_setup: cport=59132 tport=57755.
Connection from IPv4: 192.168.178.35:49267
listening for audio, control and timing on ports 6001, 6002, 6003.
Process metadata with type 73736e63, code 70626567 and length 0.
    CSeq: 38.
    Audio-Jack-Status: connected; type=analog.
    Transport: RTP/AVP/UDP;unicast;interleaved=0-1;mode=record;control_port=6002;timing_port=6003;server_port=6001.
    Session: 1.
volume:100%   repeat: off   random: off   single: off   consume: off
    Range: npt=0-.
    X-Apple-Durations: b=46; c=34; au=66; an=65; sa=16.
    CSeq: 39.
    X-Apple-Device-ID: 0x705681d6b74b.
    DACP-ID: E43FB08F535B18F4.
    Active-Remote: 2404776728.
    User-Agent: AirPlay/220.68.
    Content-Length: 0.
    CSeq: 39.
    Audio-Jack-Status: connected; type=analog.
    Audio-Latency: 88200.
    CSeq: 40.
    X-Apple-Device-ID: 0x705681d6b74b.
    DACP-ID: E43FB08F535B18F4.
    Active-Remote: 2404776728.
    User-Agent: AirPlay/220.68.
    Content-Length: 20.
    Content-Type: text/parameters.
SET_PARAMETER Content-Type:"text/parameters".
received parameters in SET_PARAMETER request.
[... 63 lines cut ...]
Process metadata with type 636f7265, code 6173646b and length 1.
Process metadata with type 636f7265, code 63617073 and length 1.
Process metadata with type 636f7265, code 6173746d and length 4.
Process metadata with type 73736e63, code 6d64656e and length 0.
    CSeq: 44.
    X-Apple-Device-ID: 0x705681d6b74b.
    DACP-ID: E43FB08F535B18F4.
    Active-Remote: 2404776728.
    User-Agent: AirPlay/220.68.
    Content-Length: 0.
    CSeq: 44.
    Audio-Jack-Status: connected; type=analog.
    Connection: close.
RTSP shutdown requested.
closing RTSP connection.
shutting down RTP thread
Audio receiver -- Server RTP thread interrupted. terminating.
Control RTP thread interrupted. terminating.
Timing RTP thread interrupted. terminating.
rtp_timing_sender thread interrupted. terminating.
Closed and terminated timer requester thread.
Timing RTP thread terminated.
Process metadata with type 73736e63, code 70656e64 and length 0.
terminating RTSP thread.
```

paviro, 2016-07-11:
This is strange now it suddenly works :/
```
culling threads.
one joined...
new RTSP connection.
    CSeq: 35.
    X-Apple-Device-ID: 0x705681d6b74b.
    Apple-Challenge: Zvirzb1/HRLyueAxYj5UGw==.
    DACP-ID: E43FB08F535B18F4.
    Active-Remote: 3404841103.
    User-Agent: AirPlay/220.68.
    Content-Length: 0.
    Apple-Response: WAvcEwoyMyKiK3kiQ4x9nMzMswhaJGpm8hppiW9oaDnfmZBRhN/BsDaoES2jVbidkqrzyFngTIxvxoy+Cph0tIoHH4tAWjDfmhv3IXtGuFHYL9c+wPZsLFtskVSsewLOG5N9tqfq8CVonHC1QeNtW0q+GqGIAPC5qn6d9+Gx5I1yyvMZThEXm8SKzuVukq0nkft0zmZtL5Qz6oBzxJv7pPJw3gp4uenV3c7duRmC6nWPqgQewki/voYHChhrydvExfwrUAdsALVditXoCXINTEgftDXV8vORUygt8n7HxIy2vL9vA88C/s3rSIMshnStsg4+Hl9QoLlC11wXfHW+sA.
    CSeq: 35.
    Audio-Jack-Status: connected; type=analog.
    Public: ANNOUNCE, SETUP, RECORD, PAUSE, FLUSH, TEARDOWN, OPTIONS, GET_PARAMETER, SET_PARAMETER.
    X-Apple-Client-Name: Fernseher.
    CSeq: 36.
    X-Apple-Device-ID: 0x705681d6b74b.
    DACP-ID: E43FB08F535B18F4.
    Active-Remote: 3404841103.
    User-Agent: AirPlay/220.68.
    Content-Length: 646.
    Content-Type: application/sdp.
Play connection from device named "Fernseher".
Play connection from user agent "AirPlay/220.68".
Process metadata with type 73736e63, code 736e616d and length 9.
    CSeq: 36.
    Audio-Jack-Status: connected; type=analog.
Process metadata with type 73736e63, code 736e7561 and length 14.
    Transport: RTP/AVP/UDP;unicast;mode=record;timing_port=55980;x-events;control_port=51331.
    CSeq: 37.
    X-Apple-Device-ID: 0x705681d6b74b.
    DACP-ID: E43FB08F535B18F4.
    Active-Remote: 3404841103.
    User-Agent: AirPlay/220.68.
    Content-Length: 0.
Active-Remote string seen: "3404841103".
DACP-ID string seen: "E43FB08F535B18F4".
Process metadata with type 73736e63, code 61637265 and length 10.
User-Agent is AirPlay; selecting the AirPlay latency of 88200 frames.
rtp_setup: cport=51331 tport=55980.Process metadata with type 73736e63, code 64616964 and length 16.

Connection from IPv4: 192.168.178.35:49361
listening for audio, control and timing on ports 6001, 6002, 6003.
Process metadata with type 73736e63, code 70626567 and length 0.
    CSeq: 37.
    Audio-Jack-Status: connected; type=analog.
    Transport: RTP/AVP/UDP;unicast;interleaved=0-1;mode=record;control_port=6002;timing_port=6003;server_port=6001.
    Session: 1.
volume:100%   repeat: off   random: off   single: off   consume: off
    Range: npt=0-.
    X-Apple-Durations: b=31; c=65; au=63; an=63; sa=17.
    CSeq: 38.
    X-Apple-Device-ID: 0x705681d6b74b.
    DACP-ID: E43FB08F535B18F4.
    Active-Remote: 3404841103.
    User-Agent: AirPlay/220.68.
    Content-Length: 0.
    CSeq: 38.
    Audio-Jack-Status: connected; type=analog.
    Audio-Latency: 88200.
    CSeq: 39.
    X-Apple-Device-ID: 0x705681d6b74b.
    DACP-ID: E43FB08F535B18F4.
    Active-Remote: 3404841103.
    User-Agent: AirPlay/220.68.
    Content-Length: 18.
    Content-Type: text/parameters.
SET_PARAMETER Content-Type:"text/parameters".
[... 358 lines cut ...]
    User-Agent: AirPlay/220.68.
    Content-Length: 0.
    CSeq: 73.
    Audio-Jack-Status: connected; type=analog.
    Public: ANNOUNCE, SETUP, RECORD, PAUSE, FLUSH, TEARDOWN, OPTIONS, GET_PARAMETER, SET_PARAMETER.
    CSeq: 74.
    X-Apple-Device-ID: 0x705681d6b74b.
    DACP-ID: E43FB08F535B18F4.
    Active-Remote: 3404841103.
    User-Agent: AirPlay/220.68.
    Content-Length: 0.
    CSeq: 74.
    Audio-Jack-Status: connected; type=analog.
    Public: ANNOUNCE, SETUP, RECORD, PAUSE, FLUSH, TEARDOWN, OPTIONS, GET_PARAMETER, SET_PARAMETER.
    CSeq: 75.
    X-Apple-Device-ID: 0x705681d6b74b.
    DACP-ID: E43FB08F535B18F4.
    Active-Remote: 3404841103.
    User-Agent: AirPlay/220.68.
    Content-Length: 0.
    CSeq: 75.
    Audio-Jack-Status: connected; type=analog.
    Public: ANNOUNCE, SETUP, RECORD, PAUSE, FLUSH, TEARDOWN, OPTIONS, GET_PARAMETER, SET_PARAMETER.

```

volumio, 2016-07-11:
Interesting...I suspect this is something related probably to your network... Did you play something different than before?

paviro, 2016-07-11:
No same thing. I will do some more testing :)

closed 2016-07-11 by paviro

paviro, 2016-07-11:
I can't seem to replicate it anymore. Really a strange bug :D