volumio/Volumio2 #541: IQAudio Digiamp muted

povserok, 2016-07-11:
If using IQaudio Digiamp, the amp is always muted. I did manage to unmute it with a sh script, but this should work automatically.

volumio, 2016-07-11:
What version are you using? Which raspberry model do you have?

povserok, 2016-07-11:
Rpi2 volumio2 rc1 hot fix

povserok, 2016-07-12:
If it is any help: I was able to unmute it with the script from IQAudio documentation (http://www.iqaudio.com/downloads/IQaudIO.pdf) and then volumio2 worked just fine. If you need any help with testing, let me know.

volumio, 2016-07-12:
It should be fixed with this
https://github.com/volumio/Volumio2/commit/08f132a0e1f4611fd39b7f4d33be07b249b9675f

However I don't recall if this was included in hotfix...

povserok, 2016-09-02:
I have just tested it again with the release 20.8.2016 and it still does not work: RPI2+Iquadio digiamp.
Everytime the power is disconnected from the RPI, the card is muted. Then i unmute it with the commands provided by IQaudio, and restart the RPI, and then it works.
For sure, this issue is still open (and annoying).

rickswan, 2016-09-18:
I have the same problem on a Rpi3 running Volumio2 0.979.  The only way I can get the card to unmute is to switch to the Rpi DAC, reboot, switch back to DigiAmp+, and reboot.  Then the mute goes off.  I tried the fixes to config.txt in the IQAudio documentation, but that doesn't work either.  This is painful.

rickswan, 2016-09-23:
Is there a fix in sight for this?  As povserok said, it is annoying and limiting the use of my system in a remote hands free environment.

closed 2017-06-02 by volumio