volumio/Volumio2 #542: RC3: Playlist Order when Adding NAS Folder

howarthcd, 2016-07-11:
When adding a NAS folder to the playlist (for instance an artist's album) the entries are added to the playlist in alphabetical order rather than in order of increasing track number as they are in Volumio 1. This is with FLAC files, I've not tried with any other file type. I also observed this behaviour in RC1 and RC2.

xipmix, 2018-09-05:
I just tested this with 2.413 (and .flac files) - this issue is still there.
If I turn on 'show track numbers' in MyMusic and clear & reload the folder into the playlist, the track numbers appear but the sort order is unchanged.
This probably should be something the user can set a preference for. What behaviour do you want to see when there are tracks from multiple albums in a playlist? Presumably the right thing to do is sort the items in each folder by track number before adding them to the play queue.

Ch1llb0, 2019-01-19:
Still having this issue with FLAC files as well... It's really annoying as the player is practically unusable like this. All albums are completely randomly shown.

xipmix, 2019-01-19:
@Ch1llb0 please can you post what version you are on for future reference?