volumio/Volumio2 #543: [testing] document how to update to latest code

xipmix, 2016-07-12:
This is more a question than an open issue but it might help with better bug reports...
What is the correct way to update a volumio install to the latest code in master?
What I have pieced together from looking over the tickets is shown below.
Happy to add whatever shakes out into the wiki.

    $ sudo bash
    # cd /
    # rm -rf ./volumio  ./data
    # git clone https://github.com/volumio/Volumio2.git volumio
    # mkdir data
    # chown -R volumio:volumio ./volumio  ./data
    # exit
    $ cd /
    $ git clone --depth 1 -b dist --single-branch https://github.com/volumio/Volumio2-UI.git /volumio/http/www
    $ cd volumio                                                                    
    $ npm install                                                                   

Relatedly, the right way to report the version one is testing? This seems reasonable...

     $ cd /volumio
     $ git remote -v; git branch -v; git log -1;

xipmix, 2016-07-19:
meh. don't you guys want help

apinner, 2016-07-19:
Sorry, i labelled it as wontfix as you said it wasn't an issue. The labels help with tracking. 

Of course any help is always very very much appreciated!

xipmix, 2016-07-24:
So I tried the procedure above and hit #556. I managed to wriggle past that one with

    $ sudo npm cache clean -f
    $ sudo npm install -g n
     /bin/n -> /lib/node_modules/n/bin/n
    /lib
    └── n@2.1.3
    $ sudo n 6.3

       install : node-v6.3.1
       mkdir : /usr/local/n/versions/node/6.3.1
       fetch : https://nodejs.org/dist/v6.3.1/node-v6.3.1-linux-armv7l.tar.gz
    ########################################################################100.0%
    installed : v6.3.1
    $ cd /volumio
    $ node -v
    v6.3.1
    $ npm install

Next hurdle: make not installed. And g++ (wants to build 'epoll'). So...

    $ sudo apt-get install build-essential

By this point I am probably pretty far from the relatively sane environment you folks are distributing so the 'npm install' would seem to be Not The Right Thing To Do. Though I seem to have got there:

    volumio@volumio:/volumio$ npm install
    > Volumio2@2.0.0-alpha4 preinstall /volumio
    > 
    Volumio2@2.0.0-alpha4 /volumio
    └─┬ onoff@1.1.1 
      └── epoll@0.1.20 
    npm WARN Volumio2@2.0.0-alpha4 No license field.

    volumio@volumio:/volumio$ git remote -v; git branch -v; git log -1
    origin  https://github.com/volumio/Volumio2.git (fetch)
    origin  https://github.com/volumio/Volumio2.git (push)
    * master 0404fc8 Proper volume mixer for hifiberry amp
    commit 0404fc82eee2e4c6505e17e2b27f085819257b2d
    Author: Volumio <info@volumio.org>
    Date:   Sat Jul 23 15:54:23 2016 +0200
    
        Proper volume mixer for hifiberry amp

And blow me down it works.

overflowed, 2016-07-25:
You should stop the volumio service instead of killing node, when you want to run it in the foreground. You can also restart this service when you do not want the foreground process.

xipmix, 2016-09-28:
This forum post shows how to use the /dev url to upgrade to the latest: https://volumio.org/forum/volumio-final-release-very-near-pre-test-needed-t4709.html. See also the procedure mentioned at the end of #714.

closed 2017-06-02 by volumio