volumio/Volumio2 #552: I can not play my radio #webradio #RC2

Theridiidae, 2016-07-16:
Adding radio, press play, and the silence of the columns.
reproducing address "http://air2.radiorecord.ru:805/rr_320"
The default radio station from the sheet are played.
Sorry for bad english.

Coolechriz, 2016-07-16:
I have the same issues when I add a radio on RC2
Streams I use http://icecast.omroep.nl/3fm-bb-mp3 and http://stream-nederland.arrow.nl//;stream/1

sawatron, 2016-07-19:
I have the same issue also when trying to play radios from Web Radio/By country(Dirble)

klokslagB, 2016-07-19:
Good news. I can play my radio on an image from buildscript. Couldn't reproduce a previous comment from me.
https://github.com/volumio/Volumio2/issues/489#issuecomment-231828642
Didn't build the image myself so I can't tell much more.

davze, 2016-07-23:
AAC streams work. MP3 do not...

sawatron, 2016-07-24:
I  believe I have found the issue. It is in the code https://github.com/volumio/Volumio2/blob/master/app/plugins/music_service/webradio/index.js#L377-L379
There is a missing:

`self.commandRouter.stateMachine.setConsumeUpdateService('mpd');`

basically the issue is when mpd load fails and  mpd add is called update is triggered and sent to statemachine however as the consumer is not set and webradio is the service gotten from the queue track it creates a mismatch issue and entry in volumio.log 

> Received update from a service different from the one supposed to be playing music. Skipping notification.

After adding the above I can listen to added radios

Coolechriz, 2016-07-24:
@sawatron Nice! But i don't understand how to build a image or add patches to my raspberry ;(
So I guess I have to wait until it is updated by Volumio

davze, 2016-07-24:
Solution of sawatron works for me.

volumio, 2016-07-24:
THank you for the feedback, we're applying as we speak!

Theridiidae, 2016-07-25:
@sawatron Tell me at what points you need to make changes.
Thank you.

sawatron, 2016-07-25:
I have o changed 
```
        .fail(function (e) {
            return self.mpdPlugin.sendMpdCommand('add "'+track.uri+'"',[]);
        })
```
to 
```
        .fail(function (e) {
            self.commandRouter.stateMachine.setConsumeUpdateService('mpd');
            return self.mpdPlugin.sendMpdCommand('add "'+track.uri+'"',[]);
        })
```
in file (starting in line 377)
https://github.com/volumio/Volumio2/blob/master/app/plugins/music_service/webradio/index.js#L377-L379

davze, 2016-07-25:
Ok, that's strange. Yesterday I was able to play my radio stations. Today they do not play anymore.

sawatron, 2016-07-25:
I do not know if this is the same but after several hours of playing webradio it stops and I cannot play it any more (I cannot play anyhing really)I have to restart raspberry and it works again

5AMsan, 2016-08-10:
Same issue here. Same mp3 stream playing flawlessly when played from mpd.

closed 2017-06-02 by volumio