volumio/Volumio2 #553: Missing volume up/down buttons on mobile interface

Coolechriz, 2016-07-16:
Hello, 
On the mobile interface I miss the volume up and down buttons enormous, the use of the slider will sometime bug making it sound suddenly goes much too hard. Is there a possibility to add these buttons to the interface?
Thanks!

ps, sorry for bad English, i'm Dutch

TeenTin, 2016-07-16:
Agree.  The volume slider is hard to use.

amalic, 2016-07-20:
please don't forget to not show any volume controls if volume change is disabled in the settings -> https://github.com/volumio/Volumio2/issues/547

heis2201, 2016-07-25:
When using the WebUI in Safari on my iphone, the volume slider doesn't even work: Not able to change volume with the slider.

Coolechriz, 2016-07-26:
@heis2201 on my iPhone whit Google Chrome it is working but hard to use

apinner, 2016-07-29:
Added here https://github.com/volumio/Volumio2-UI/commit/0c471fd164b413d6346831cfc48d357668f7f18e

closed 2016-07-29 by apinner