volumio/Volumio2 #554: DEV-console PLAY-command not working

amalic, 2016-07-16:
How to reproduce:

1. Play a song in Volumio2
2. Go to http://<your-volumio2-ip>/dev 
3. Press "PAUSE" button
4. Press "PLAY" button --> nothing happens

NemoAsura, 2016-07-17:
This is an error in the dev.js file.
Line 11 shout be corrected from:
`function() {emitEvent('play', '');}`
to:
`function() {emitEvent('play', null);}`

wga22, 2016-12-21:
This same error is affecting /volumio/app/plugins/system_controller/volumio_command_line_client/commands/playback.js

xipmix, 2016-12-21:
Can you explain _why_ this works? What consumes the emitted Event? 

The reason I ask is there are lmore places this construct appears, e.g. line 12 of dev.js uses the same construct for 'pause' but that apparently works just fine(?) The emitEvent function in socket-io wants a data item as the second element so perhaps a real ```null``` value is better than a blank string.  The suggested change to playback.js makes more sense as ```null``` will be returned as the ```data``` part for any issued command.

closed 2017-06-03 by volumio