volumio/Volumio2 #555: Add Google Play Music support

amalic, 2016-07-16:
Since I don't use Spotify, but Google Play Music i am desperately missing Google Play Music support. I own a Chromecast Audio but I don't like the sound of it. I preffer the sound of my DurioPro-DAC.

I did some researching. Unfortunately I just started node.js development because of Volumio2, and I don't think I will be soon ready to implement it myself. Would be great if someone could take up this task.

Here are two possible tools which could be used for integration

- GMusicProxy ([Website](http://gmusicproxy.net/), [GitHub](https://github.com/diraimondo/gmusicproxy))
- Mopidy ([Website](https://www.mopidy.com/), [GitHub](https://github.com/mopidy/mopidy))

Mopidy seems to me to be a replacement for mpd, GMusicProxy more like an addon which can be connected to mpd.

amalic, 2016-07-16:
GMusicProxy seems quite promising. I got it running on the Volumio2 image. I was able to play Music form Google Play music.

e.g.  
First I starded GMusicProxy as daemon. gmusicproxy.cfg contains my email-address and google password. This could also be provided via the command line.
`GMusicProxy --daemon --config gmusicproxy.cfg`

Then i used an a slightly modified sample from the GmuicProxy Instructions
`mpc clear; curl -s 'http://localhost:9999/get_new_station_by_search?type=artist&artist=Radiohead&num_tracks=100' |   grep -v ^# | while read url; do mpc add "$url"; done; mpc play`

It worked very well. The Audio-quality via the DurioPro-DAC was better than via ChromecastAudio.

amalic, 2016-07-18:
I was thinking how this could work quite simple. You could support playing only user defined playlists from google play music.
`curl -s 'http://localhost:9999//get_all_playlists'`
This lists all user defined playlists with their **id**

When the user selects a playlist you could get the m3u playlist for the specific item
`mpc stop; mpc clear; curl -s 'http://localhost:9999/get_playlist?id=<selected playlist id>' > /var/lib/mpd/playlists/volumio_gmusic.m3u'; mpc load volumio_gmusic; mpc play;`

Fetching a playlist from google music takes time.

amalic, 2016-07-21:
This here might also be interresting for you guys. Node.js play music library
--> https://github.com/jamon/playmusic

apinner, 2016-07-21:
This is something that could be added as a plugin like Spotify quite easily i would imagine.

amalic, 2016-07-21:
Sounds promising. I will try to test the NodeJS API and let you know!

renich, 2016-08-30:
I am interested in this as well.

death2all110, 2017-08-13:
Any updates on this?

XaF, 2017-09-30:
Would be interested in this too.

sla89, 2017-10-07:
Its under GPL v3 - is that a problem?

davidblumntcgeo, 2018-03-25:
Also interested!

nklmilojevic, 2018-04-11:
Any updates on this?

xniala, 2018-07-21:
Interested too.
Any chance to see it one day ?

clickwir, 2018-11-13:
This would be great, Google Music is the only streaming service I use. It works really well, but it doesn't have as many people using it so it doesn't get the 3rd party love like some other services. Please have this as an option in Volumio!

tukaibat, 2019-07-06:
Would like to get started with integrating Chromecast support in this. This would enable Volumio to catch music from sources like Google Play Music, Spotify etc. Almost in the manner like it supports Apple Airplay. Can anyone point me to the code files, where I should start looking? I'm looking at this Google Developers page: https://developers.google.com/cast/docs/caf_receiver/basic.

trtompkins, 2020-08-04:
Since Google has announced that Play Music is being sunset in favor of YouTube Music, any challenges with incorporating this?  The YouTube plugin seems to work pretty well now (after having to update /data/plugins/music_service/youtube/package.json file with "ytdl-core": "^3.2.0" in the "dependencies section"

Thoughts???