volumio/Volumio2 #556: Unable to run volumio with Node.js v6.2.2

amalic, 2016-07-16:
```
volumio@volumio:~/volumio$ node index.js
info: -------------------------------------------
info: -----            Volumio2              ----
info: -------------------------------------------
info: -----          System startup          ----
info: -------------------------------------------
info: Plugin folders cleanup
info: Scanning into folder /home/volumio/volumio/app/plugins/
info: Scanning category audio_interface
info: Scanning category miscellanea
info: Scanning category music_service
info: Scanning category plugins.json
info: Scanning category system_controller
info: Scanning category user_interface
info: Scanning into folder /data/plugins/
info: Plugin folders cleanup completed
info: Loading plugins from folder /home/volumio/volumio/app/plugins/
info: Loading plugins from folder /data/plugins/
info: Loading plugin "system"...
info: Loading plugin "appearance"...
/home/volumio/volumio/node_modules/bindings/bindings.js:91
  throw err
  ^

Error: Could not locate the bindings file. Tried:
 → /home/volumio/volumio/node_modules/lwip/build/lwip_image.node
 → /home/volumio/volumio/node_modules/lwip/build/Debug/lwip_image.node
 → /home/volumio/volumio/node_modules/lwip/build/Release/lwip_image.node
 → /home/volumio/volumio/node_modules/lwip/out/Debug/lwip_image.node
 → /home/volumio/volumio/node_modules/lwip/Debug/lwip_image.node
 → /home/volumio/volumio/node_modules/lwip/out/Release/lwip_image.node
 → /home/volumio/volumio/node_modules/lwip/Release/lwip_image.node
 → /home/volumio/volumio/node_modules/lwip/build/default/lwip_image.node
 → /home/volumio/volumio/node_modules/lwip/compiled/6.2.2/linux/arm/lwip_image.node
    at bindings (/home/volumio/volumio/node_modules/bindings/bindings.js:88:9)
    at /home/volumio/volumio/node_modules/lwip/lib/Image.js:3:41
    at Object.<anonymous> (/home/volumio/volumio/node_modules/lwip/lib/Image.js:16:3)
    at Module._compile (module.js:541:32)
    at Object.Module._extensions..js (module.js:550:10)
    at Module.load (module.js:458:32)
    at tryModuleLoad (module.js:417:12)
    at Function.Module._load (module.js:409:3)
    at Module.require (module.js:468:17)
    at require (internal/module.js:20:19)
volumio@volumio:~/volumio$ node -v
v6.2.2
```

volumio, 2016-07-16:
Modules require node 6.3

closed 2016-07-16 by volumio

amalic, 2016-07-16:
you should update the README file because it says

> Volumio works with 5.5.0 version of NodeJS. Reports of working\not working node version are appreciated!