volumio/Volumio2 #558: getState returns pause when radio channel is playing

tomatpasser, 2016-07-17:
Reproduce: Play a radio station

Pause by using the node script

`var io=require('socket.io-client');`
`var socket= io.connect('http://localhost:3000');`
`socket.emit('pause');`

(player pauses correctly)

Start by using the node script

`var io=require('socket.io-client');`
`var socket= io.connect('http://localhost:3000');`
`socket.emit('play');`

(player starts again correctly)

Ask what the player is doing:

`var io=require('socket.io-client');`
`var socket= io.connect('http://localhost:3000');`
`socket.emit('getState', '');`
`socket.once('pushState', function (state) {`
`		console.log('State: ' + state.status);`
`});`

Output:
`State: pause`

xipmix, 2017-04-04:
I can reproduce this on 2.129 (rpi).

tomatpasser, 2017-04-04:
Indeed seems to still be an issue. Had to make a workaround to make GPIO-Buttons plugin work with radio playback

xipmix, 2017-04-04:
There are lots of problems here as far as I can tell, that start well before getting into play/pause changes.
Making the changes shown in #1116 modifies the playback behaviour a little, I get these messages in the log
```
info: [1491312868529] Received update from a service different from the one supposed to be playing music. Skipping notification.Current webradio Received mpd
```
Instead of a ```info: STATE SERVICE``` line, so I clearly haven't got things right there.

Not quite sure where to go with this next. I don't think ```app/statemachine.js``` is going to do the right things until the parsing issues are sorted out, in particular the setting of the ```service``` and ```trackType``` fields.

closed 2017-06-02 by volumio