volumio/Volumio2 #560: Please add replace and play to conextmenu when selecting folder/title

amalic, 2016-07-18:
I'm missing a comfort feature from Volumio2. "Replace and Play" which replaces the current playlist by the selected folder/title and plays the selected title.

Problem description:
I've been playing before "The Top 100 Masterpieces of Classical Music". Now I'm playing "Kruder & Dorfmeister - The K&D Sessions". I have to do a lot of scrolling to select a specific title in the playlist.

apinner, 2016-07-19:
IIRC just clicking Play from the menu does exactly that.

amalic, 2016-07-19:
not really. it adds the new song/album to the current playlist.

heis2201, 2016-07-25:
.... and create a function **"Replace & Play"**. Thanks!

apinner, 2016-07-29:
Done here https://github.com/volumio/Volumio2/issues/584#issuecomment-235334921

closed 2016-07-29 by apinner