volumio/Volumio2 #562: Mopidy instead of MPD?

paviro, 2016-07-18:
Ever thought about using [Mopidy](https://www.mopidy.com/) instead of MPD? Seems to have some great modules!

amalic, 2016-07-18:
You mean things like Google Play Music support? 
see --> https://github.com/volumio/Volumio2/issues/555

paviro, 2016-07-18:
SoundCloud, Podcast-iTunes, YouTube, InternetArchive, also [Moped](https://docs.mopidy.com/en/latest/ext/web/#mopidy-moped) looks really nice with sorting by Album etc. and large images etc.

amalic, 2016-07-18:
I think Moped is a UI that comes with Mopidy.

amalic, 2016-07-18:
Here's a list what is available out of the box for the volumio2 image
```
volumio@volumio:~$ apt-cache search mopidy
mopidy - music server with support for MPD/HTTP clients
mopidy-alsamixer - Mopidy extension for extension for ALSA volume control
mopidy-beets - Mopidy extension for playing music from Beets' web plugin
mopidy-dirble - Mopidy extension for browsing Dirble's radio station directory
mopidy-doc - music server with support for MPD/HTTP clients - documentation
mopidy-mpris - Mopidy extension for controlling playback through MPRIS
mopidy-scrobbler - Mopidy extension for scrobbling music to Last.fm
mopidy-soundcloud - Mopidy extension for playing music from SoundCloud
mopidy-tunein - Mopidy extension for playing music from TuneIn
mopidy-local-sqlite - Mopidy extension for keeping your local library in SQLite
mopidy-dleyna - Mopidy extension for playing music from Digital Media Servers
mopidy-internetarchive - Mopidy extension for playing music from the Internet Archive
mopidy-podcast - Mopidy extension for searching and browsing podcasts
mopidy-somafm - Mopidy extension for playing music from SomaFM
mopidy-podcast-itunes - Mopidy extension for searching and browsing iTunes podcasts
mopidy-youtube - Mopidy extension for playing music from YouTube
mopidy-podcast-gpodder - Mopidy extension for searching and browsing gpodder.net podcasts
mopidy-spotify - Mopidy extension for playing music from Spotify
mopidy-spotify-tunigo - Mopidy extension for providing the browse feature of Spotify
```

paviro, 2016-07-18:
Don`t have the time right now busy with https://github.com/MichMich/MagicMirror stuff was just curios :)

toggm, 2020-04-27:
I successfully installed mopidy together with mopidy-dleyna and other modules with the following steps:

**1. SSH to the image**
**2. Install mopidy from mopidy repository**
```
wget -q -O - https://apt.mopidy.com/mopidy.gpg | sudo apt-key add -
sudo wget -q -O /etc/apt/sources.list.d/mopidy.list https://apt.mopidy.com/jessie.list
sudo apt update && sudo apt install mopidy mopidy-dleyna mopiy-mpd
```
_do not install mopidy-dleyna from pip as this needs the latest version of mopidy which is not available for jessie_

**3. Disable MPD / Enable Mopidy**
```
sudo systemctl disable mpd
sudo systemctl stop mpd
sudo systemctl enable mopidy
sudo systemctl start mopidy
```
**4. Apply patch to Volumio**
Apply the following path to your volumio installation:
https://github.com/volumio/Volumio2/compare/master...toggm:dev/volumio_with_mopidy?expand=1

**5. Adjust mopidy configuration**
* Expose mopidy http publicly to be able to connect with the mopidy mobile client
* Configure audio interface correctly. My configuration based on a justboom AMP hat is ```[audio]
output = alsasink device=hw:1,0
```
* Adjust local and files location to the /media directory to be able to access attached USB devices ```
[local]
media_dir = /media

[file]
media_dirs = /media
```

Some additional notes:
* Be aware that the patch might slightly change based on the source code version you have locally as especially the pidof part was changed lastly:https://github.com/volumio/Volumio2/commit/0503d4b60062bddd8d5c96b1f3224e9d1e401508#diff-2965d5982753ac5ebe2a4ba4f826b3a2
* This patch also enables browse/play capability based on mopidy/dleyna backend. Search and browsing works nice
* If you change audio interface settings, MPD service gets started automatically

Things to improve:
* Integrate mopidy configuration to the UI
* search of albums or artists with empty string returns an empty result. This is currently used to initialize albums and artists collection and is therefore empty.

volumio, 2020-04-28:
What are the advantages of mopidfy against mpd?

toggm, 2020-04-28:
On one side mopidy support additional extensions. On the other side the missing Upnp search functionality which works well in mopidy-dleyna. First tried to get the Upnp search functionality working (from the existing dev branch), without success and as I already use mopidy on my server it was quite obvious to stick to a working solution.

volumio, 2020-04-28:
Which extensions does it support?

toggm, 2020-04-28:
For me the main reason was the missing search functionality if your music library is backed by a upnp/DLNA server. Mopidy seems to be more active than MPD so there might even more be coming:
https://mopidy.com/ext/