volumio/Volumio2 #567: Two problems

hardy8, 2016-07-20:
I have two problems with mij PI 3 and Volumio2.

issue 1:
I use a diy pcm1794 DAC with i2s connection, and it works fine on OpenElec with all sample rates up to 192Khz.
 On Volumio only mp3 and FLAC with 96Khz sample works well, but there is now sound from WAV, FLAC 44.1Khz, 48,2Khz and 192Khz.(only short glitch). 
I use the Hiffi-berry inerface. 

issue 2:
Wifi will not work on Volumio, but I have try de latest, but older RuneAudio and here it works well (has also the DAC problem).
I can't find any bad configuration.

Hardy.

amalic, 2016-07-20:
have you tried checking if volumio has set the correct settings?

can you do "cat /etc/modules" on the rpi and post your settings here?

i'm also curious about your project. (schematics, website, link to kit?)

amalic, 2016-07-20:
any more infos on your wifi problem?

this might help -> https://www.maker.io/en/blogs/raspberry-pi-3-how-to-configure-wi-fi-and-bluetooth/03fcd2a252914350938d8c5471cf3b63

hardy8, 2016-07-20:
Tanks Alexander for your reply
Hier is my modules folder.I 'm develope a compleet Preamp with an arduino include de Dac, but now have got the PI 3 and want to use the Dac also with it. The Dac works perfect with OpenElec, but not with volumio, I think it is an timing problem, but I don't no how to fix this problem.Also, why works wifi with the old runeaudio and not with the volumio2, I can't compare it with each other, volumio2 has change the folder structure 
Hardy

Date: Wed, 20 Jul 2016 10:35:29 -0700

  
  

 		 	   		  
# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.

##snd-bcm2835
snd_soc_core
snd_soc_bcm2708_i2s
bcm2708_dmaengine
snd_soc_pcm1794a
snd_soc_rpi_dac

amalic, 2016-07-21:
Since you use a PCM1794 chip IMHO you should try selecting RPI-DAC in your settings which uses the same chip.
-> http://www.tjaekel.com/T-DAC/schematics.html

Your /etc/modules file looks like you're already using the RPI-DAC settings. The link also says "The DAC is a TI PCM1794A. It is used with I2S format which supports 24bit (only) and 48..192 KHz.".

Maybe you need to set up the sample rate conversion (16Bit --> 24Bit). I guess OpenElac does this autmatically. 

I know there was a setting for Volumio. Not sure about Volumio2 --> will check when I get home.

amalic, 2016-07-23:
my etc modules contains only one line (except of the comments)

`snd_bcm2835`

i'm using a durio pro I2C sound card

volumio, 2016-07-23:
I  know that durio sound needs their own drivers... We're going to integrate them... Could you point me to any reference for this?

hardy8, 2016-07-23:
Thanks, I try this and let you now.
Hardy

Date: Sat, 23 Jul 2016 04:51:05 -0700

hardy8, 2016-07-23:
Thanks for reply, my Dac is basically the same.Let me now when I can try it.
regards Hardy

Date: Sat, 23 Jul 2016 06:53:22 -0700

hardy8, 2016-07-29:
I have try it, but without succes.
Hardy

Date: Sat, 23 Jul 2016 04:51:05 -0700

amalic, 2016-08-02:
I don't think durio is the same as your DIY DAC.

AFAIK Durio uses the PCM5102A dac chip, whereas your dac is utilizing the PCM1794 chip.

Your DAC can be compared to the RPI DAC.

hardy8, 2016-08-02:
You are right,  I have seen it, but its shout also work, but it doesn't.
 
 
 
 
-------Oorspronkelijk bericht-------
 
Van: Alexander Malic
Datum: 08/02/16 06:02:51
Aan: volumio/Volumio2
Cc: hardy8;  Author
Onderwerp: Re: [volumio/Volumio2] Two problems (#567)
 
I don't think durio is the same as your DIY DAC.
AFAIK Durio uses the PCM5102A dac chip, whereas your dac is utilizing the
PCM1794 chip.
Your DAC can be compared to the RPI DAC.

hardy8, 2016-08-09:
Thanks for the driver update in Volumio , but it changed nothing , the " Hifi Berry Dac " and the "Generic I2S Dac " drivers either work but still only MP3 and HR 96/192 and does not FLAC or WAV 44.1.
I thought the "R -Pi Dac " driver should actually works , but here I have no sound.                                  The circuit of  this  Dac is about the same as I use here .                                               Hardy
Date: Sat, 23 Jul 2016 06:53:22 -0700

hardy8, 2016-08-09:
Here are the circuit.http://www.tjaekel.com/T-DAC/schematics.html

Date: Sat, 23 Jul 2016 06:53:22 -0700

closed 2017-06-02 by volumio