volumio/Volumio2 #569: Support for Amazon Prime Music

apinner, 2016-07-21:
I don't think there is an official API yet, but seems someone has written a kodi plugin that could probably be reverse engineered. 

https://github.com/pitpompej/kodi_plugins/

Amazon Prime seems to be growing really quickly, especially with so many services for one price. So it would be nice to be ahead of the curve.

amalic, 2016-07-23:
very cool!

Shuro, 2016-09-29:
Totally agree we need that. :+1:

erichegit, 2016-10-03:
Amazone Music Integration would be amazing.
I have the feeling - there are to less Plugins available - Volumio is such a cool Player - I use it a lot - but from Android Devices there is no way to stream easy,...
Please implement a Amazon Music Plugin :)

jerryrard, 2016-10-19:
That would be awesome! I also have a Spotify premier account, but would love if they had one for Amazon as well.

woodgeek, 2017-01-20:
Is anyone working on an Amazon music plugin?

volumio, 2017-01-21:
AFAIK no one is working on it...

chsims1, 2017-01-21:
@volumio Do you think we need a more detailed and newbie friendly guide to producing plugins.  This one and playlist imports (for example) are really prime contenders, but no one is doing it, presumably because they consider it beyond their capabilities.  I did sit down and consider playlists, but the info is not enough.

sla89, 2017-02-27:
Actually I started developing a plugin.

lukinsj, 2017-09-26:
Has there been any progress on a plugin for Amazon Music?

rzo1, 2018-12-30:
Any updates? :) Ist someone working on it?

marcomayer, 2019-12-30:
Any news on this, now with Amazon Music HD being available it would be really interesting!