volumio/Volumio2 #571: Wiki: Setting Up a Dev Environment

overflowed, 2016-07-22:
Think would be helpful if you could add that you need to have a make and a compiler to build some of the dependend node modules.
So I would suggest to add to the Gettting Started section:
6. Login via ssh to the Volumio and install build-essential by
sudo apt-get install build-essential

Update and and Prepare DEV folder could benefit from the hint that not all modules build their binary modules when issuing a npm install in /volumio and need another npm install within that module dir.

xipmix, 2016-07-24:
Link to the wiki page: [Setting-Up-a-Dev-Environment](https://github.com/volumio/Volumio2/wiki/Setting-Up-a-Dev-Environment). 

Could you please list the binary modules that need that extra 'npm install' ?
Greetings from #543.

overflowed, 2016-07-25:
lwip for sure, rest I can't remember but 2 or 3 other as well

xipmix, 2016-07-25:
Tried lwip and got a bunch of 'error parsing json'. But eventually it did build the C++, much the same as it did before - so perhaps we are seeing different behaviours.

Could you describe what it was that tipped you off the binary modules were not getting built? Looking back over the build log from my attempt (ie running grep make)  it looks like it tried to build binary modules for - epoll, inotify, libxmljs, lwip, mdns. All of those have a

    COPY Release/<foo>.node

at the end of the build phase, so I assume that means they were properly built & fully installed, but (ovbiously) I'm a node noob...

overflowed, 2016-07-25:
I went to /volumuio and run npm install, this failed first time at epoll, because I was missing build-essential. I installd build-essential and gave npm install another try it looked like it finished without any errors but when I tried to run the app with node index.js lwip was complaining about missing binary modules so I went into the lwip folder run another npm install and the modules were built without any errors. This I repeated for some other packages.

xipmix, 2016-07-26:
Ok, thanks for your help. I gave it a try

    cd /volumio; service volumio stop; rm -rf /data/*; node index.js

The only thing that seemed to be missing was the multiroom plugin, definitely didn't see anything about lwip in the log output, and no indications of missing libraries. I think I'll try to reflash RC2+Hotfix and work forward from there.

closed 2017-06-02 by volumio