volumio/Volumio2 #572: MPD Client Issues

Xtgskc, 2016-07-22:
Various issues with mpd clients and  (iOS - MPoD, MpaD, Auremo, from now 'Client'),

Client playlist queue only displays track currently being played (not full queue) if the track was selected from Volumio GUI (either form a playlist or manually added eg play a single track or an album).

If a track is selected from a Client it effectively plays in the background without them displaying at all in the Volumio GUI. Volumio does not display any information (track name, progress etc) in the 'Queue' or 'Playback' pages of the GUI . Cannot pause / skip / stop playback from the GUI.  Clients display the correct information (track, progress etc).

Volume control is not working correctly from Client: 

Changes in volume from the Client are not updating the volume 'dial' in the GUI (did not work in 1.55 but would be nice). Auremo shows changes made to volume by MPod/PaD but Volumio does not show changes on its volume slider (in 1.55 changes made in the Volumio GUI were displayed on Auremo).

Raspi 3.5mm Jack, 'PCM' volume control.
Clients seem to control volume in a linear manner disregarding the 'Volume Curve Mode'setting from within playback.

Raspi Hifiberry DAC+, 'Digital' volume control.
Clients have no control in either 'Linear' or 'Natural' mode. Clients are sending correct data to Volumio (if changed on MPaD/PoD change is shown in Auremo).

Playlists:
Playlists stored from Volumio GUI are not visible from Client.
Playlists stored from Client are not visible from Volumio GUI.

amalic, 2016-07-23:
seems that volumio2 is managing the playlist itself.

how to reproduce
play an album with multiple tracks on volumio2
type `mpc clear` in a ssh-console (if mpc is not installed, do a `sudo apt-get install mpc` first). instead of having an empty playlist it plays the next song.

volumio, 2016-07-23:
Yes, volumio handles payback queue and playlist in a different way. To see them in mpd clients we need to progress development on a mpd interface plugin here:
https://github.com/volumio/Volumio2/tree/master/app/plugins/user_interface/mpdemulation

Is anyone willing to help?

amalic, 2016-07-23:
i would help out, but my node.js skills are not yet good enough.

Xtgskc, 2016-07-24:
I too would love to help but alas my skills in this area leave everything to be desired. :-(

arthurlutz, 2016-11-07:
I am a bit surprised that these bugs have been identified for a number of months and very little effort has been put into this. Is this a marginal use case scenario for volumio ? My understanding from documents such as the architecture described in https://volumio.org/technical-overview/ was that the MPD/MPC ecosystem was an important feature for volumio. 

I can fully understand that this is not a priority (as stated in the comments, you invite for some contributions on that matter), but in that case, I would suggest removing MPD from the core, have it as a plugin and remove it from the features that you advertise in the documents, thus reducing the hopes in that a potential user can have of that functionality (or explain that it's an either/or type of functionality).

benoliver999, 2016-11-07:
From what I gather Volumio does actually still use MPD (correct me if I'm wrong), it's just that the web-interface does its own thing with regards to playlists etc.

I use the current version with lots of different MPD clients and it works great, and the status syncs across all of them. It's a shame that it doesn't work with the web-ui, but I don't think removing MPD is the solution.

If I remember correctly, the 'old' version actually did work as an MPD client, so it _is_ a shame to see the feature disappear. But... like everyone else... I don't really have the skills to make it work myself, so I shouldn't moan too much.

volumio, 2016-11-20:
Duplicated, goes on here: https://github.com/volumio/Volumio2/issues/494

closed 2016-11-20 by volumio