volumio/Volumio2 #574: Support for Pandora Radio

heis2201, 2016-07-22:
Hi, 

anyone capable of writing a plugin to integrate Pandora streaming to Volumio?
Thanks

skikirkwood, 2016-07-30:
This would be challenging, as Pandora only makes their API available to qualified partners:

https://www.quora.com/Do-Spotify-or-Pandora-provide-APIs-to-fetch-user-data-e-g-who-listened-to-what-in-a-given-period-of-time-for-non-commercial-use

There is an unofficial API but apparently Pandora changes their API frequently to break 3rd parties who are not official partners.  They should, like Spotify, open up the API to anyone.

heis2201, 2016-08-01:
Oh, if that's the case, let's drop the idea until Pandora decides to let more people use their service...

closed 2016-08-01 by heis2201